BeamCtrl/Airiana #4: sending data

phk-r, 2020-06-30:
Hi!
Is there any way to modify this to send the data over something like MQTT aswell and not only to a web page?

Best regards

BeamCtrl, 2020-06-30:
nothing is impossible, I can probably add it to my backlog if you are interested in such functionality.
I have not yet done any MQTT implementations but it seems to be too much of a hassle. What System air unit are you using?

//Daniel

phk-r, 2020-06-30:
> nothing is impossible, I can probably add it to my backlog if you are interested in such functionality.
> I have not yet done any MQTT implementations but it seems to be too much of a hassle. What Systemair unit are you using?
> 
> //Daniel

If not MQTT then maybe some other way to forward the data to another instance. In my case home assistant. 
I'm using a VSR 300.

BeamCtrl, 2020-06-30:
I have already implemented an HTTP-sensor for homeassistant. , it forwards humidity and temperature inside / outside to HA.
//Daniel

BeamCtrl, 2020-06-30:
to get it working you will need to run the core with option homeAss in the command line and have your bearer token added to the ha-httpsensor.py file

phk-r, 2020-07-04:
> to get it working you will need to run the core with option homeAss in the command line and have your bearer token added to the ha-httpsensor.py file

Ok, not sure how to fix that but I'll try to figure it out next week :) 
Is it possible to send commands via HA to change fan speed etc.?

BeamCtrl, 2020-07-05:
The homeAssistant integrations are sketchy at best,  I have not done more than try to get it working, to enable the http sensor, you will need to if you installed using the install-script, edit the service /etc/systemd/system/airiana.service to include homeAss as a cmd line opt. you will have to add the bearer token which is generated by homeAssistant in the ha-httpsensor.py file on var. ”Bearer”. To set fanspeed etc, use the RUST-component to do http-requests, look in ./public/buttons.html file for the proper request of type, ” ”http://airiana.local:8000/command?xx”.

closed 2020-07-23 by BeamCtrl

BeamCtrl, 2020-08-07:
update, i found that there is a dependency missing on the home ass side, you will need to do a "sudo pip install requests" to get it working, it has been fixed in the latest commit.

BeamCtrl, 2020-08-07:
the configuration of HomeAssistant http sensor has changed update the ha-token file with your server address port and bearer token to get it working