BeamCtrl/Airiana #5: Systemair IAM module support /Modbus TCP

kromosomX, 2020-08-06:
Hi, 

it's a very interesting project. Can you change the underlying modbus implementation to one that supports Modbus TCP, like https://github.com/AdvancedClimateSystems/umodbus/ or https://github.com/riptideio/pymodbus? This would open up a lot of possibilities for connection including Systemair Internet Acess Module that supports Modbus TCP. 

Best regards, 

Tomaz

BeamCtrl, 2020-08-06:
Hello, nice to see you take an interest in this project. I would guess an update to use an TCP implementation would be quite straight forward by updating the request class to be confiugrable TCP or RTU, but  I’m not sure why you would need to. Running this setup on a Rpi is more or less equvalent to an IAM but with added functionality. What Functionality are you looking for by going thru the IAM?
 Best regards /Daniel

kromosomX, 2020-08-06:
Hi, it's not the functionality. It's convenience. I already have the IAM module connected. I also have the RPI but it's used as an interface for the Vaillant heat pump. For the time being i'll just integrate the Systemair into Home Assistant via modbus2mqtt and use the MQTT Climate integration and set up some autiomations/scripts to tweak it via Home Assistant. I will 'borrow' some of your ideas. I am already controling some of the functionality (Kitchen Hood and Central Vacuum Modes) of the Systemair via digital inputs.

The Systemair automation  is not working as i would expect by default. For example I have to set the temperature very low (like 12C) to stop the recuperation of heat in the summer (during the day, during the night it goes into FreeCooling mode).  

 Also if I use on demand automation in connection to humidity and the humidity is to high it increases the ventilation even if the outside air is hoter and more humid.

BeamCtrl, 2020-08-07:
I did a fast implementation of pyModbusTCP to be used as request handler, I have not been able to test it since I have no IAM. give it a shot and tell me if it has alot of buggs ;).
//daniel, 
note that you will need to edit the /etc/systemd/system/airiana.service file to add "TCP" as an cmd line arg. and restart the service. can be done with ."./restart" in the airiana directory after a systemctl "daemon-reload"

enjoy // Daniel

kromosomX, 2020-08-08:
Hi, 
thank you for the TCP implemantation. 
I made  a few changes (commented out some modbus RTU related lines and fixed a few things in request) and now i can get  airiana-core.py running directly. There are some read and write errors comming up and temperature values are off by one decimal but it is runnning. 

I have a VTR300, so i presume that savecair should be set to True. Right?

I'll play a little more with it and report back the changes that I make. 

Thanks again. I also have to figure out how to  issue pull requests so the changes can be merged back after I clean them up a little.

BeamCtrl, 2020-08-08:
it should automatically set to true if it gets a positive response on a SaveCair adress, the savecair uses a different address space compared to the legacy systemair units

kromosomX, 2020-08-09:
I disabled the update for the time being as I am going on a short trip tomorrow and I won't be back before Thursday. 

I managed to get everything up and running. Now the automatic SaveCair recognition is working (i moved setting the modbus client variable for TCP in the airiana-core.py also)

If i set the debug command line my pi freezes after a couple of minutes (at least  when i run it from the command line/ didn't try it as a service). 

If you do any changes in the meanwhile i will merge my changes with yours manually when i get back and then i'll try to figure out how to do pull requests. 

I presume that i should keep the VTR300 in manual mode, correct? Are there any specific settings that i should set on the VTR or NOT set for airiana to work?

I have a ton of questions regarding the features of airiana  (shower mode, preemptive cooling,...) but that will wait until I come back. 

Thank you for your help. BTW What unit do you use?

BeamCtrl, 2020-08-09:
Great fun to hear you are getting it working, to have it not go into conflict with the existing features what I have done on my  VTR300/savecair is disable free cooling, humidity recovery.
If you toggle the auto on/off you will turn all automated features off, and the airiana unit will only read from the unit.

I will have to do a write up of how the features work and what is needed by the user to get it working, a sort of manual that is clear ;).

when strange stuff happens you can always check the err log in ./RAM/err for any hints on why debug causes crashes.

anyway great to have you onboard to get the TCP -> IAM working, its hard to test without one =).
best regards// Daniel.
have a great trip!

kromosomX, 2020-08-09:
Hi, 
you probably don't get temperatures over 30C in Västerås very often? Hehe.The hack for the exchanger works only only until 30C outside temperature. So I was scratching my head today (31C) as why the exchanger is not working. When the outside temperature fell below 30C everything started working as expected. If I enable Cold Recovery and set the temperature below outside temperature the exchanger works. Would using this technique interfere with other functionality? I'll rewrite the whole logic for starting and stoping the exchanger on Savacair units if it doesn't interfere with other functionality. 

Best regards, 

Tomaž

BeamCtrl, 2020-08-09:
It is true not that often ;), how ever, the hack to turn on the exchanger is a legacy from my old VR400 unit, I’m actually considering removing the whole block and just relying on the internal heat recovery logic. Do you feel ot works well?

BeamCtrl, 2020-08-15:
I’ve been running my savecair unit without the exchange hack for cooling recovery and it seems the unit handles it quite well. I will remove the force exchanger section shortly.

kromosomX, 2020-08-15:
Yeah, i think the dafult cooling recovery feature of savecair works OK. But the free cooling feature of savecair does not (too many limitations). Airiana works much better for cooling.

BeamCtrl, 2020-12-19:
implemented additions in PR 5.

closed 2020-12-19 by BeamCtrl