BeamCtrl/Airiana #8: Electrical power calculation / VTR300

kromosomX, 2020-08-16:
I think the electric power calculation is wrong for the VTR300.
In airiana i get  a value of 325W when the electrical meter hooked up to VTR300 is showing 170W. In VTR300 specifications Pmax is 179W (without the heater).

BeamCtrl, 2020-08-16:
i have added a scaling factor ,0.57 for savecair units, you can try to test some different speed combinations on your unit to see i power consumption matches, this formula that is used was derrived from my old vr400 unit and the motor characteristics may be a bit off but should generally be a fair match.

kromosomX, 2020-08-16:
This is much better. Here are the values i get now:
Fan setting, RPM-S, RMP-E, Airinana EP, Measured EP
High; 3240; 3240; 185; 172
Norm: 1860; 1860; 37; 42
Low: 780; 780; 3; 15

The exchanger only adds 1W to the consumption, when the fan in running low. Funny thing is that when the fan is High the consumption falls 1-2W when the exchanger starts.

BeamCtrl, 2020-08-16:
ok so I decided to do it properly, added new equation, power = 3.404*exp(0.001*SFfan_rpm) + 3.404*exp(0.001*SFfan_rpm) ..
matches quite well here however I can only read the power draw to -1 decimal place.
//Daniel

closed 2020-08-16 by BeamCtrl