BeamCtrl/Airiana #10: Connection problems

danielkaldheim, 2020-08-19:
Hi, 
awesome project!

I'm having some trouble connecting to my HVAC (Systemair VR 400 DCV/B R). I have tried changing baud rate and slave id, but the response from the unit is not correct.

Here's a dump of RAM/err and RAM/out:
https://gist.github.com/danielkaldheim/cca07903f6734031c1fda6edd6703354

What do you think may be the problem?
Many thanks!

BeamCtrl, 2020-08-19:
Hi, hope I can help. You are experiencing no communication, I see that it is trying to do com on ttyACM0, I guess you have done the changes yourself in the airiana-core to to use ttyACM0? what hardware are you using to connect to the VR400 unit? it looks from the output that you are not getting any kind of response on the serial line. 
fault finding tips:
1.check that A/B cables are'nt switched. if you have an rx/tx led on your recieved it will be steady always on if cables are swapped.
2. is the serialport correct, is your transciver connected to ttyACM0?
3. baudrate config, 19200 no pairity 1 stopbit, slave id 1

danielkaldheim, 2020-08-19:
I have tried different modules: I have change the unit variable in airiana-core.py to work with [CC-USB-RS485-150U](https://www.ebay.com/itm/CC-USB-RS485-150U-EN-MPPT-Solar-Epsolar-Connected-To-PC-Communication-Cable-AM/253067353581?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3Dce8491f1276c43d089a89956ae6cabd5%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dnone%26sd%3D253067353581%26itm%3D253067353581%26pmt%3D0%26noa%3D1%26pg%3D2380057%26brand%3DUnbranded&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3Af5d08b4f-e244-11ea-b6e6-74dbd1801c92%7Cparentrq%3A07db31991740aa465e7df92ffffff9aa%7Ciid%3A1) ttyACM0, and the wiring to match the specification.

I have also tried [FT232 USB](https://cdn-shop.adafruit.com/datasheets/FT232_Model.pdf) (ttyUSB0) connected with [SparkFun Transceiver Breakout - RS-485](https://www.sparkfun.com/products/10124) where only the transmission led is blinking.

I have tried switching A/B cables with no luck.

On my systemair controller there is a RJ45 terminal marked modbus, and by the [specs](https://github.com/danielkaldheim/my-public-notes/blob/master/Projects/Systemair/pdfs/12314-Modbus%20for%20Residential%20units_D24810_User_manual%20(A005).pdf) on page 2 it looks like A/B lines is on pin 5 and 4.

Here's some more photos: https://github.com/danielkaldheim/my-public-notes/blob/master/Projects/Systemair/index.md

BeamCtrl, 2020-08-19:
Yes, you seem to have done your homework and the dongle is transmitting 👍🏻. I’d try to change the address to 0 on the vr400 and keep it on 1 in the code. Did you install termination resistors?

danielkaldheim, 2020-08-20:
Yeah, I have been off and on this project for years.

The SparkFun breakout have a termination resistor, not sure about the CC-USB-RS485-150U. I'm not able to change the address to 0, only from 1 to 247. But I'll have another try tonight with different address setups. 
I also have borrowed a [USB-RS485-WE-1800-BT](https://no.mouser.com/ProductDetail/895-USBRS485WE1800BT?r=895-USBRS485WE1800BT) that I'll try out tonight.

Thanks for your suggestions!

BeamCtrl, 2020-08-20:
I'd double check and test to switch A/B connectors just to test, I have made several installations on vr400 units and it is always the case that I need to swap them. electrically it looks good so I see no reason for it not to work. just to make sure, you are connecting to the RJ45 in the middle of the controller board labeled MODBUS? if you update to the latest commit and checkout the save2.py file you can test your linksettings....

danielkaldheim, 2020-08-20:
Awesome! Thank you! I'll check that out!

I just tested with the borrowed USB-RS485 cable and this works fine, so I have no clue why the other two cables / boards don't work. But now I can confirm that the setup works, that I have located the correct pins from controller board, and of course your software works!

closed 2020-08-20 by danielkaldheim