BeamCtrl/Airiana #13: How to run on bare metal?

devployment, 2020-12-28:
The `./install.py` and other parts seem to be specific for Raspberry. At least that's what I think after briefly scanning the code. 

In my case I'd like to run Airiana in my virtualization (Proxmox). By that it would run in a Ubuntu 20.04 container (LXC).

Could you give me some pointers about what is specific for Raspberry and if it is feasible trying to make the code running in a non Raspberry environment? If so, I will try to make it run and create a PR if I succeed.

BeamCtrl, 2020-12-28:
Cool =),. check the dependencies in install.py, and get them,  check airiana-core.py from around line 75 where it tests for avail char-device files to choose from to do the communication.  there is room for improvement here; its a qucik and dirty way to support multiple com options; and add your RS485-device. 

hopefully that should be it, for autostart there is a systemd service file in ./systemfiles/  it also has all availible input arguments specified. 

depending on how you want the unit write to disk, the install script sets up ram-drives for logging and file generation to the webserver, in ./public/. choose how you want the files in ./RAM/ to be accessible in ./public for access to the web-server, I have sym-links made in the install script.

if you want auto updates which pull down this repo and updates when I deem a new stable release is avail. it adds a crontab check on install, ref the install script for/if you like to have this enabled.

give me a shout here if she gives you grievance...
Good luck .//Daniel

BeamCtrl, 2020-12-28:
It seems that python2 is no longer packaged with ubuntu 20 I guess you will need to install python2 and make sure that python and python3 are correctly symlinked as well

closed 2021-04-06 by BeamCtrl