BeamCtrl/Airiana #16: ESP8266/ESP32

Ex3mXX, 2021-01-07:
Like to see this implemented in ESPHome.
ESP8266/ESP32 with a Modbus RTU.

BeamCtrl, 2021-01-08:
I would say it’s too much work, and logging updateing etc will be a problem,
My tip would be to use the ESP as a modbu TCP/RTU bridge, Airiana has support for using a TCP reciever as target and modbus bridge libs are avail for the ESPs. You can then run the logic processing on any machine on your network.

/Daniel

Ex3mXX, 2021-01-27:
In my case I just like to get all the information to HASS and for there take care of the logging and logic.
So by just setting up a ESP as a modbu TCP/RTU bridge with ESPHome and send the data to HASS would make my day!
Anyhow, will try out your solution with a Rasp. 

Thanks
Patrik

closed 2021-04-06 by BeamCtrl