BeamCtrl/Airiana #17: New install fails to use RAM disk

johanberglin, 2021-05-29:
**Describe the bug**
I get RAM disk errors after installing and webserver does not start

**To Reproduce**
Steps to reproduce the behavior:
I have a PI zero W 1.1
I useed raspberry pi imager to put latest vanilla raspberry pi image on sdcard (both light and normal version tested)
updated so that it connected to wifi 
after initial boot via putty i ran these commands
sudo apt-get update 
sudo apt-get upgrade
sudo apt-get install git
git clone https://github.com/BeamCtrl/Airiana
cd Airiana
sudo python ./install.py

**Expected behavior**
Airiana starts properly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

BeamCtrl, 2021-05-29:
Thank you for this, found a series of static paths that are not updated properly during install. Will update shortly.

BeamCtrl, 2021-05-30:
pushed fixes, now the install script will update static paths to match during installation. from `690edfc`

closed 2021-05-30 by BeamCtrl