BeamCtrl/Airiana #21: New install - webserver does not start and RAM disk errors

overas, 2022-01-27:
**Describe the bug**
I get RAM disk errors after installing and webserver does not start. Maybe the same issue as in #17.

**To Reproduce**
I have a raspberry pi 3
I downloaded Raspberry Pi Os Lite Legacy from here: https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2021-12-02/2021-12-02-raspios-buster-armhf-lite.zip and write the pi-image to sd-card.
After initial boot I enabled ssh-server with raspi-config and changed timezone. 

Then I ran these commands:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git
git clone https://github.com/BeamCtrl/Airiana
cd Airiana
sudo python ./install.py

After reboot I edit /etc/systemd/system/airiana.service and added "TCP" to commandline option.
Then I edited the ipconfig file with correct ip and port for my Systemair Internet Access Module.

**Expected behavior**
Airiana webserver starts properly

**Screenshots**
pi@airiana:~/Airiana $ ./watch
[image]
[image]

**Additional context**
Add any other context about the problem here.
Run "sudo python ./install.py"
`pi@raspberrypi:~/Airiana $ sudo python ./install.py
Installing the AirianaCores
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  gir1.2-glib-2.0 javascript-common libexpat1-dev libgirepository-1.0-1 libjs-jquery libjs-sphinxdoc libjs-underscore libpython-all-dev libpython-dev
  libpython2-dev libpython2.7 libpython2.7-dev python-all python-all-dev python-asn1crypto python-cffi-backend python-configparser python-crypto
  python-cryptography python-dbus python-dev python-entrypoints python-enum34 python-gi python-ipaddress python-keyring python-keyrings.alt
  python-pip-whl python-secretstorage python-setuptools python-six python-wheel python-xdg python2-dev python2.7-dev
Suggested packages:
  apache2 | lighttpd | httpd python-crypto-doc python-cryptography-doc python-cryptography-vectors python-dbus-dbg python-dbus-doc python-enum34-doc
  python-gi-cairo gnome-keyring libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-gdata python-keyczar python-secretstorage-doc python-setuptools-doc
The following NEW packages will be installed:
  gir1.2-glib-2.0 javascript-common libexpat1-dev libgirepository-1.0-1 libjs-jquery libjs-sphinxdoc libjs-underscore libpython-all-dev libpython-dev
  libpython2-dev libpython2.7 libpython2.7-dev python-all python-all-dev python-asn1crypto python-cffi-backend python-configparser python-crypto
  python-cryptography python-dbus python-dev python-entrypoints python-enum34 python-gi python-ipaddress python-keyring python-keyrings.alt python-pip
  python-pip-whl python-secretstorage python-setuptools python-six python-wheel python-xdg python2-dev python2.7-dev
0 upgraded, 36 newly installed, 0 to remove and 0 not upgraded.
Need to get 36.3 MB of archives.
After this operation, 62.9 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf python-pip-whl all 18.1-5+rpt1 [1,591 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf python-pip all 18.1-5+rpt1 [215 kB]
Get:2 http://raspbian.trivini.no/raspbian buster/main armhf libgirepository-1.0-1 armhf 1.58.3-2 [78.8 kB]
Get:4 http://raspbian.trivini.no/raspbian buster/main armhf gir1.2-glib-2.0 armhf 1.58.3-2 [143 kB]
Get:5 http://raspbian.trivini.no/raspbian buster/main armhf javascript-common all 11 [6,120 B]
Get:6 http://raspbian.trivini.no/raspbian buster/main armhf libexpat1-dev armhf 2.2.6-2+deb10u1 [127 kB]
Get:7 http://raspbian.trivini.no/raspbian buster/main armhf libjs-jquery all 3.3.1~dfsg-3+deb10u1 [332 kB]
Get:8 http://raspbian.trivini.no/raspbian buster/main armhf libjs-underscore all 1.9.1~dfsg-1+deb10u1 [100.0 kB]
Get:9 http://raspbian.trivini.no/raspbian buster/main armhf libjs-sphinxdoc all 1.8.4-1 [95.9 kB]
Get:10 http://raspbian.trivini.no/raspbian buster/main armhf libpython2.7 armhf 2.7.16-2+deb10u1 [873 kB]
Get:11 http://raspbian.trivini.no/raspbian buster/main armhf libpython2.7-dev armhf 2.7.16-2+deb10u1 [30.9 MB]
Get:14 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpython-all-dev armhf 2.7.16-1 [1,064 B]
Get:15 http://raspbian.raspberrypi.org/raspbian buster/main armhf python-all armhf 2.7.16-1 [1,044 B]
Get:17 http://raspbian.raspberrypi.org/raspbian buster/main armhf python2-dev armhf 2.7.16-1 [1,216 B]
Get:18 http://raspbian.raspberrypi.org/raspbian buster/main armhf python-dev armhf 2.7.16-1 [1,192 B]
Get:19 http://raspbian.raspberrypi.org/raspbian buster/main armhf python-all-dev armhf 2.7.16-1 [1,068 B]
[... 532 lines cut ...]
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Enabling uart
Setting up Ram drive
setup autostart for airiana.service
Created symlink /etc/systemd/system/multi-user.target.wants/airiana.service → /etc/systemd/system/airiana.service.
setup autostart for controller.service
Created symlink /etc/systemd/system/multi-user.target.wants/controller.service → /etc/systemd/system/controller.service.
adding boot cmdline config
Copy hosts file
Touch data.log
setup symlinkts between RAM and ./public
ln: failed to create symbolic link 'out.txt': File exists
ln: failed to create symbolic link 'history.png': File exists
chown: cannot access '../RAM/': No such file or directory
chown: cannot access '../RAM/*': No such file or directory
setup auto update from airiana repo.
no crontab for pi
Installation completed, reboot in 15 sec
`

BeamCtrl, 2022-01-29:
will be addressed as part of #22, note that to read the error log you need to `cat ./RAM/err`, it is not an executable file but an human readable of error and command printouts.

BeamCtrl, 2022-10-09:
closed as of python3 integration

closed 2022-10-09 by BeamCtrl