BeamCtrl/Airiana #22: New install - Python install issue with Raspberry Pi image "Bullseye"

overas, 2022-01-27:
**Describe the bug**
Airiana Python requirements do not install with install.py. I also get RAM disk errors.

**To Reproduce**
I have a raspberry pi 3
I downloaded Raspberry Pi Os from here: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf-lite.zip and write the pi-image to sd-card.
After initial boot I enabled ssh-server with raspi-config and changed timezone.

Then I ran these commands:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git
git clone https://github.com/BeamCtrl/Airiana
cd Airiana
sudo python ./install.py

**Expected behavior**
Airiana webserver starts properly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

**First try:** 
pi@raspberrypi:~/Airiana $ sudo python ./install.py
  File "/home/pi/Airiana/./install.py", line 16
    print "Installing the AirianaCores"
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Installing the AirianaCores")?

I found out the error message is because install.py use Python 3. I installed Python 2
pi@raspberrypi:~/Airiana $ sudo apt install python2
pi@airiana:~ $ python --version
Python 2.7.18

**Second try with Python 2:**
pi@raspberrypi:~/Airiana $ sudo python2 ./install.py
Installing the AirianaCores
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
The following additional packages will be installed:
  libexpat1-dev libpython2-dev libpython2.7 libpython2.7-dev python-is-python2 python2-dev python2.7-dev
The following packages will be REMOVED:
  python-is-python3
The following NEW packages will be installed:
  libexpat1-dev libpython2-dev libpython2.7 libpython2.7-dev python-dev-is-python2 python-is-python2 python2-dev python2.7-dev
0 upgraded, 8 newly installed, 1 to remove and 0 not upgraded.
Need to get 3,272 kB of archives.
After this operation, 12.8 MB of additional disk space will be used.
Get:1 http://raspbian.trivini.no/raspbian bullseye/main armhf libexpat1-dev armhf 2.2.10-2 [121 kB]
Get:2 http://raspbian.trivini.no/raspbian bullseye/main armhf libpython2.7 armhf 2.7.18-8 [865 kB]
Get:3 http://raspbian.trivini.no/raspbian bullseye/main armhf libpython2.7-dev armhf 2.7.18-8 [1,967 kB]
Get:4 http://raspbian.trivini.no/raspbian bullseye/main armhf libpython2-dev armhf 2.7.18-3 [21.3 kB]
Get:5 http://raspbian.trivini.no/raspbian bullseye/main armhf python-is-python2 all 2.7.18-9 [2,940 B]
[... 175 lines cut ...]
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for dbus (1.12.20-2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Enabling uart
Setting up Ram drive
setup autostart for airiana.service
Created symlink /etc/systemd/system/multi-user.target.wants/airiana.service → /etc/systemd/system/airiana.service.
setup autostart for controller.service
Created symlink /etc/systemd/system/multi-user.target.wants/controller.service → /etc/systemd/system/controller.service.
adding boot cmdline config
Copy hosts file
Touch data.log
setup symlinkts between RAM and ./public
ln: failed to create symbolic link 'out.txt': File exists
ln: failed to create symbolic link 'history.png': File exists
chown: cannot access '../RAM/': No such file or directory
chown: cannot access '../RAM/*': No such file or directory
setup auto update from airiana repo.
no crontab for pi
Installation completed, reboot in 15 sec

BeamCtrl, 2022-01-29:
Started work on a full python3 migration, thanks for taking the time to add issues about the fact that the os baseline now has left py2.7 behind.

overas, 2022-01-30:
Thanks for taking your time to fix the migration and for a great software. 
The documentation for installing Airiana are confusing:

> -Update the latlong.json file with your position to enable the forcast for your position.

I says the user must update latlong.json file, but I do not find the file. 

> -edit the ip-config file.
I think it should be renamed to ipconfig file in the Airiana folder.

BeamCtrl, 2022-02-16:
If you'd like to test stuff out while I debug. there is a branch avail called python3-migration. This should work fully on python3 only systems. =)

overas, 2022-02-17:
Great. I am going to install python3-migration this weekend.

BeamCtrl, 2022-02-19:
seems it does not create the files due to a permission issue, I updated the service files in the repo to execute as root user instead of pi, you can manually edit the user in file `/etc/systemd/system/airiana.service` and `/etc/systemd/system/controller.service`. 
manual execution of the core is also available for testing by using `sudo ./airiana-core.py` unitll I can figure out how to close the gap  with regards to not having the ram drive mount with root ownership.

overas, 2022-02-20:
Thanks. I going to try execute as root.

overas, 2022-03-05:
Thanks for your work to migrate to Python3. This is a status update with the newest git update. I think you know about this, but starting as root gives errors:

`pi@airiana:~/Airiana-python3-migration $ sudo ./airiana-core.py

Traceback (most recent call last):
  File "/home/pi/Airiana-python3-migration/./airiana-core.py", line 20, in <module>
    os.chdir(path)
NotADirectoryError: [Errno 20] Not a directory: '/home/pi/Airiana-python3-migration/airiana-core.py'
`

`pi@airiana:~/Airiana-python3-migration $ sudo service airiana status
● airiana.service - Airiana Core executive
     Loaded: loaded (/etc/systemd/system/airiana.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-03-05 15:26:26 CET; 2s ago
    Process: 868 ExecStart=/usr/bin/python3 /home/pi/airiana/airiana-core.py daemon ping humidity TCP (code=exited, status=2)
   Main PID: 868 (code=exited, status=2)
        CPU: 77ms

Mar 05 15:26:26 airiana systemd[1]: airiana.service: Scheduled restart job, restart counter is at 5.
Mar 05 15:26:26 airiana systemd[1]: Stopped Airiana Core executive.
Mar 05 15:26:26 airiana systemd[1]: airiana.service: Start request repeated too quickly.
Mar 05 15:26:26 airiana systemd[1]: airiana.service: Failed with result 'exit-code'.
Mar 05 15:26:26 airiana systemd[1]: Failed to start Airiana Core executive.
`

BeamCtrl, 2022-03-05:
Hi, actually I was unaware, thought I had recovered from strange path handling backwards compat issues when removing the use of pathlib, I was misstaken, A fix is up now which runs on my unit, give it a spin. =)

overas, 2022-03-08:
Airiana webpage is now running. 
First the the webpage was not running. I got some warning about path not found in line 3 in ip-replace.sh. I renamed the downloaded git zipfolder from "Airiana-python3-migration" to "airiana". Then the webpage worked.
I did the install again "sudo python ./install.py" in the airiana folder. The RAM folder was created without error. So it sees the correct name worked for the install script also.

[image]

[image]

[image]

[image]

The webpage is only for local access. How can I get remote access to the webpage from my LAN?
*No access allowed outside of local domain.

BeamCtrl, 2022-03-08:
Hi, good to see we're making some progress. It looks like there is some kind of communications issue with the IAM, I found a bug where a zero len string response is considered as a passing communication test that is now fixed as of commit 24d3d325. Can you verify that the IAM module can accept connections and is responding?  A note on using the log files; they are not executable use "`cat RAM/err"` and "`cat RAM/error_rate`" to see content.

I added a simple RTU/TCP test to request.py to simplify testing, "sudo ./request.py now give a printout of the initial setup and subsequent selftest of communication to indicate if it is to use the savecair address space or the old legacy one. commit c5ba1538

note that if you manually change the name of the directory you will need to remove the systemd files  run the install script  again with the clean argument "sudo python3 install.py clean" to reset some static paths (its in my backlog to make all paths relative).

overas, 2022-03-10:
Started with a fresh install raspbian desktop. Downloading Airiana-python3-migration and unzipping. The folder is not renamed. Installed airiana with sudo ./request.py
Edited ipconfig with {"ip":"10.0.0.107", "port":505} and airiana.service
with "TCP". 

pi@airiana:~/Airiana-python3-migration $ ./restart
● airiana.service - Airiana Core executive
     Loaded: loaded (/etc/systemd/system/airiana.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-03-10 18:27:46 CET; 88ms ago
   Main PID: 3000
      Tasks: 1 (limit: 4915)
        CPU: 83ms
     CGroup: /system.slice/airiana.service

Mar 10 18:27:46 airiana systemd[1]: Started Airiana Core executive.
Mar 10 18:27:46 airiana python3[3000]: /usr/bin/python3: can't open file '/home/pi/airiana/airiana-core.py': [Errno 2] No such file or directory
Mar 10 18:27:46 airiana systemd[1]: airiana.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 10 18:27:46 airiana systemd[1]: airiana.service: Failed with result 'exit-code'.
sh: 1: disown: not found
pi@airiana:~/Airiana-python3-migration $

The airiana service is not starting. Starting with sudo ./airiana-core.py
pi@airiana:~/Airiana-python3-migration $ sudo ./airiana-core.py
./ip-replace.sh: line 3: cd: /home/pi/airiana/public/: No such file or directory
Communication started on device Serial0;
Trying to Run on host: airiana , for 60sec;
Reporting system start;
Using RTU backend;
request object created:  RTU ;

Running Comm test;
pi@airiana:~/Airiana-python3-migration $ ^C
pi@airiana:~/Airiana-python3-migration $ cat ./RAM/err

Using /dev/serial0
read error high rate,
            possible no comms with unit error rate over 90%
read error high rate,
            possible no comms with unit error rate over 90%
read error high rate,
            possible no comms with unit error rate over 90%
read error high rate,
            possible no comms with unit error rate over 90%
read error high rate,
            possible no comms with unit error rate over 90%
pi@airiana:~/Airiana-python3-migration $ 

pi@airiana:~/Airiana-python3-migration $ cat RAM/error_rate
-6.132610392555334e-08 0.01
pi@airiana:~/Airiana-python3-migration $ 

pi@airiana:~/Airiana-python3-migration $ sudo python request.py
Communication started on device Serial0;
Using RTU backend;
request object created:  RTU ;

Running Comm test;
pi@airiana:~/Airiana-python3-migration $

Do not now if it helps, but I use TCP port 505 with Home Assistant:
modbus:
  - type: tcp
    host: 10.0.0.107
    port: 502
    name: "VTR500"
Here is a link to the IAM access module I use. https://makitech.no/ventilasjon/boligaggregater/tilbehor-til-boligaggregater/internett-tilgangsmodul-for-SAVECair-aggregater#intro

BeamCtrl, 2022-03-10:
First I'd like to thank you for helping me with debugging this, it turned out to be quite the handful to migrate, but necessary 
I'm a little bit confused on what exactly happened here, seems you start with RTU and with incorrect paths. anything saying RTU should be bad in your case. we want the request class to talk TCP with the IAM.
paths are made correct when you exec the `install.py` script
 I suggest this install procedure

- `git clone https://github.com/BeamCtrl/Airiana.git`
-` cd Airiana `
- `git checkout python3-migration`
- `sudo python3 install.py clean`  _( this command when used with clean removes the systemd files and reinstalls them after path replacements to match install path you will however need to re edit and add the TCP argument to core execution)_
- edit "/etc/systemd/system/airiana.service" add TCP to arguments
- add your TCP config, please check if port 502 or 505 is the correct one, saw you used both above.
- test with `./request.py`, you should see 

> 
> pi@airiana:~/airiana $ ./request.py 
> Communication started on device Serial0;
> Using RTU backend;
> request object created:  RTU ;
> 
> Running Comm test;
> Testing Non-savecair address 101:0;
> Testing savecair address 12543:222;
> Comm test:  True ;
> 
> Using TCP backend
> 
> Reading ip configuration file for IAM access {"ip":"localhost", "port":505}
> 
> request object created:  TCP ;
> 
> Running Comm test;
> TCP read error on addrs: 101
> Testing Non-savecair address 101:;
> TCP read error on addrs: 12543
> Testing savecair address 12543:;
> Comm test:  False ;
> 

But with the TCP  Comm test passing, I have it connected thru RTU.

-` ./restart`

overas, 2022-03-11:
Thanks for your kind words. I started with a fresh raspbian desktop install and then after configure and update the operativsystem followed your procedure for installing Airiana.
Then I tested the communication with IAM (ip 10.0.0.107) with the tcp port 505 and 502 with request.py. I sorry to say that both test was not sucessfull.

pi@airiana:~/Airiana $ sudo python3 request.py 
Communication started on device Serial0;
Using RTU backend;
request object created:  RTU ;

Running Comm test;
Traceback (most recent call last):
  File "/home/pi/Airiana/request.py", line 168, in modbusregister
    self.response = self.client.read_register(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 480, in read_register
    returnvalue = self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1322, in _perform_command
    response_bytes = self._communicate(request_bytes, number_of_bytes_to_read)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1490, in _communicate
    raise NoResponseError("No communication with the instrument (no answer)")
minimalmodbus.NoResponseError: No communication with the instrument (no answer)

---Repeats many times: Traceback (most recent call last), then it logs:

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Airiana/request.py", line 168, in modbusregister
    self.response = self.client.read_register(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 480, in read_register
    returnvalue = self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1322, in _perform_command
    response_bytes = self._communicate(request_bytes, number_of_bytes_to_read)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1460, in _communicate
    answer = self.serial.read(number_of_bytes_to_read)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 577, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Airiana/request.py", line 246, in <module>
    req.setup(unit, "RTU")
  File "/home/pi/Airiana/request.py", line 76, in setup
    test = self.comm_test()
  File "/home/pi/Airiana/request.py", line 82, in comm_test
    self.modbusregister(101, 0)  # Read non savecair flow address
  File "/home/pi/Airiana/request.py", line 178, in modbusregister
    self.modbusregister(address, decimals)
  File "/home/pi/Airiana/request.py", line 178, in modbusregister
    self.modbusregister(address, decimals)
  File "/home/pi/Airiana/request.py", line 178, in modbusregister
    self.modbusregister(address, decimals)
  [Previous line repeated 41 more times]
  File "/home/pi/Airiana/request.py", line 168, in modbusregister
    self.response = self.client.read_register(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 480, in read_register
    returnvalue = self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1322, in _perform_command
    response_bytes = self._communicate(request_bytes, number_of_bytes_to_read)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1399, in _communicate
    self.serial.reset_input_buffer()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 661, in reset_input_buffer
    termios.tcflush(self.fd, termios.TCIFLUSH)
termios.error: (5, 'Input/output error')
[... 57 lines cut ...]
   Main PID: 10188 (python3)
      Tasks: 9 (limit: 4915)
        CPU: 1.169s
     CGroup: /system.slice/airiana.service
             ├─10188 /usr/bin/python3 /home/pi/Airiana/airiana-core.py daemon ping humidity TCP
             ├─10219 /bin/sh -c ./forcast2.0.py pressure
             ├─10220 /usr/bin/python3 ./forcast2.0.py pressure
             ├─10221 sudo python3 ./public/httpsrv.py
             ├─10222 python3 ./public/httpsrv.py
             ├─10223 sh -c ./geoloc.py
             ├─10224 /usr/bin/python3 ./geoloc.py
             ├─10225 /bin/sh -c curl 'https://api.ipify.org?format=json'
             └─10226 curl https://api.ipify.org?format=json

Mar 11 19:07:26 airiana airiana-core.py[10188]: file is /home/pi/Airiana/airiana-core.py
Mar 11 19:07:26 airiana airiana-core.py[10188]: changing to /home/pi/Airiana/
Mar 11 19:07:26 airiana sudo[10201]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/pkill -f httpsrv.py
Mar 11 19:07:26 airiana sudo[10201]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 11 19:07:26 airiana sudo[10201]: pam_unix(sudo:session): session closed for user root
Mar 11 19:07:26 airiana sudo[10215]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/pkill -f httpsrv2.py
Mar 11 19:07:26 airiana sudo[10215]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 11 19:07:27 airiana sudo[10215]: pam_unix(sudo:session): session closed for user root
Mar 11 19:07:27 airiana sudo[10221]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/python3 ./public/httpsrv.py
Mar 11 19:07:27 airiana sudo[10221]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
sh: 1: disown: not found

overas, 2022-03-11:
I think I was a little quick with the feedback. The communication with the IAM works now after changing the port to 502. When using tcp 505 the airiana service  fails.

pi@airiana:~/Airiana $ cat ./RAM/request.log 
Testing Non-savecair address 101:0
Testing savecair address 12543:222
Request object Passed communications test.
pi@airiana:~/Airiana $ 

The webinterface (airiana controller) is active, but when I push the buttons "Fanspeed Low", High or Normal nothing happens. I have tried the other buttons and nothink happens with them either.

pi@airiana:~/Airiana $ service controller status
● controller.service - Airiana Controller TCP(http) --> UDP server
     Loaded: loaded (/etc/systemd/system/controller.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-03-11 18:35:39 CET; 1h 43min ago
   Main PID: 362 (python3)
      Tasks: 2 (limit: 4915)
        CPU: 1.586s
     CGroup: /system.slice/controller.service
             └─362 /usr/bin/python3 /home/pi/Airiana/public/controller.py

Mar 11 18:35:39 airiana systemd[1]: Started Airiana Controller TCP(http) --> UDP server.
Mar 11 19:00:00 airiana systemd[1]: /etc/systemd/system/controller.service:4: Unknown key name 'Type' in section '>
Mar 11 19:00:00 airiana sys

pi@airiana:~/Airiana $ service airiana status
● airiana.service - Airiana Core executive
     Loaded: loaded (/etc/systemd/system/airiana.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-03-11 20:20:22 CET; 2s ago
   Main PID: 23462 (python3)
      Tasks: 4 (limit: 4915)
        CPU: 1.404s
     CGroup: /system.slice/airiana.service
             ├─23462 /usr/bin/python3 /home/pi/Airiana/airiana-core.py daemon ping humidity TCP
             ├─23497 sudo python3 ./public/httpsrv.py
             └─23498 python3 ./public/httpsrv.py

Mar 11 20:20:23 airiana airiana-core.py[23462]: file is /home/pi/Airiana/airiana-core.py
Mar 11 20:20:23 airiana airiana-core.py[23462]: changing to /home/pi/Airiana/
Mar 11 20:20:23 airiana sudo[23479]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/pkill -f https>
Mar 11 20:20:23 airiana sudo[23479]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 11 20:20:23 airiana sudo[23479]: pam_unix(sudo:session): session closed for user root
Mar 11 20:20:23 airiana sudo[23493]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/pkill -f https>
Mar 11 20:20:23 airiana sudo[23493]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 11 20:20:23 airiana sudo[23493]: pam_unix(sudo:session): session closed for user root
Mar 11 20:20:23 airiana sudo[23497]:     root : PWD=/home/pi/Airiana ; USER=root ; COMMAND=/usr/bin/python3 ./publ>
Mar 11 20:20:23 airiana sudo[23497]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
pi@airiana:~/Airiana $ 

pi@airiana:~/Airiana $ cat ./RAM/err
                                 Dload  Upload   Total   Spent    Left  Speed
100    21  100    21    0     0     57      0 --:--:-- --:--:-- --:--:--    57
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   937  100   937    0     0   9464      0 --:--:-- --:--:-- --:--:--  9561
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: api.apility.net
Traceback (most recent call last):
  File "/home/pi/Airiana/airiana-core.py", line 1963, in <module>
    device.get_forcast()
  File "/home/pi/Airiana/airiana-core.py", line 1665, in get_forcast
    self.forcast[2] = float(forcast[1])
ValueError: could not convert string to float: 'None\n'

Using /dev/serial0
System started	Fri Mar 11 20:20:39 2022
Traceback (most recent call last):
  File "/home/pi/Airiana/./forcast2.0.py", line 56, in <module>
    f = open("latlong.json")
FileNotFoundError: [Errno 2] No such file or directory: 'latlong.json'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    21  100    21    0     0     52      0 --:--:-- --:--:-- --:--:--    52
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   937  100   937    0     0   9760      0 --:--:-- --:--:-- --:--:--  9760
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: api.apility.net
Savecair unit set

overas, 2022-03-12:
Output from ./watch and ./RAM/out
[image]

pi@airiana:~/Airiana $ cat ./RAM/out
Reporting system start;
Using TCP backend
Reading ip configuration file for IAM access {"ip":"10.0.0.107", "port":502}
request object created:  TCP ;

Running Comm test;
Testing Non-savecair address 101:0;
Testing savecair address 12543:224;
Comm test:  True ;

value error for pressure
TCP read error on addrs: 12543
Going in for first PASS;
First PASS;
Updating fanspeeds;
pi@airiana:~/Airiana $ 

Airiana has not created a latlong.json file. 

I start up ./cmd.py and use the menu
Sat Mar 12 11:12:14 2022        CTRL-C to exit,
1: Toggle Auto Monitoring        6:
2: Toggle fanspeed               7: Toggle pressure diff
3: Print all device attributes   8: Run fans for 120min at Max
4: Display link settings         9:
5: show/update values           10: Toggle shower mode
                enter commands:
^Cpi@airiana:~/Airiana $ 

Nothing happens with the systemair VTR500 fans when I choose 10: "Toggle shower mode" or 8: "Run fans for 120 min at Max"

BeamCtrl, 2022-03-12:
We’re closing in… you have confirmed comms. It is crashibg because it tries to geolocate and failing. As such it does not manage go get the forecast for your pos. Pull the latest, I added a new source of geo location data that should fins you.

overas, 2022-03-12:
The fix for geolocate solved the crashing of the Airiana service. I have now a latlong.json with latlong location 
{"lat":"58.34046","long":"8.59337"}
and Airiana read the data from the vtr500

[image]
 
The buttons for Fanspeed Low, Normal and High and Forced Ventilation do not work.  When I push them the fan do not changed speed. Auto on/off works. I do not know if Winter/Summer mode works because the Airiana do not give any output i the terminal window. 

pi@airiana:~/Airiana $ cat ./RAM/err 
to: None	Sat Mar 12 17:38:26 2022
('127.0.0.1', 45002):99 at	Sat Mar 12 17:38:58 2022
Changing fanspeed to:3 		Sat Mar 12 17:38:58 2022
Incorrectly set fanspeed 2 to 3 	Sat Mar 12 17:38:59 2022
('127.0.0.1', 39404):98 at	Sat Mar 12 17:39:33 2022
('127.0.0.1', 52031):0 at	Sat Mar 12 17:39:55 2022
cycle exchanger to: None	Sat Mar 12 17:39:55 2022
('127.0.0.1', 57376):0 at	Sat Mar 12 17:41:01 2022
cycle exchanger to: None	Sat Mar 12 17:41:01 2022
Traceback (most recent call last):
  File "/home/pi/Airiana/./grapher.py", line 136, in <module>
    ax.set_xlim(min(time[-day:-1]), max(time[-day:-1]))
ValueError: min() arg is an empty sequence

Using /dev/serial0
System started	Sat Mar 12 17:48:16 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
  File "/home/pi/Airiana/./grapher.py", line 133, in <module>
    ax.set_ylim(int(min(inlet)) - 1, int(max(extract + inlet + exhaust + supply + outside)) + 2)
ValueError: min() arg is an empty sequence
Pressure difference set to: 10%	Sat Mar 12 17:48:27 2022

Using /dev/serial0
System started	Sat Mar 12 17:50:53 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
  File "/home/pi/Airiana/./grapher.py", line 133, in <module>
    ax.set_ylim(int(min(inlet)) - 1, int(max(extract + inlet + exhaust + supply + outside)) + 2)
ValueError: min() arg is an empty sequence
Pressure difference set to: 10%	Sat Mar 12 17:51:04 2022

Using /dev/serial0
System started	Sat Mar 12 17:53:54 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
  File "/home/pi/Airiana/./grapher.py", line 133, in <module>
    ax.set_ylim(int(min(inlet)) - 1, int(max(extract + inlet + exhaust + supply + outside)) + 2)
ValueError: min() arg is an empty sequence
Pressure difference set to: 10%	Sat Mar 12 17:54:03 2022

Using /dev/serial0
System started	Sat Mar 12 17:54:16 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
[... 42 lines cut ...]
Changing fanspeed to:3 		Sat Mar 12 18:26:34 2022
Incorrectly set fanspeed 2 to 3 	Sat Mar 12 18:26:34 2022
('127.0.0.1', 43404):1 at	Sat Mar 12 18:26:53 2022
Changing fanspeed to:1 		Sat Mar 12 18:26:57 2022
Incorrectly set fanspeed 2 to 1 	Sat Mar 12 18:26:58 2022
Dynamic fanspeed 1 with RH	Sat Mar 12 18:26:58 2022
('127.0.0.1', 37899):0 at	Sat Mar 12 18:27:00 2022
cycle exchanger to: None	Sat Mar 12 18:27:00 2022
('127.0.0.1', 47938):1 at	Sat Mar 12 18:27:12 2022
pi@airiana:~/Airiana $ 

When using ./cmd.py and choose "2: Toggle fanspeed" or "10: Toggle shower mode" or "8: Run fans for 120 min at Max" nothing happens. 
When I choose "3: Print all device attributes" the program crash with output:

break

Traceback (most recent call last):
  File "/home/pi/Airiana/./cmd.py", line 33, in <module>
    raw_input("press enter to resume") 
NameError: name 'raw_input' is not defined
pi@airiana:~/Airiana $

The home assistant integration works, but I must change the port from port":8234 to port":8123 in the "ha-token" file. I get three sensor in Home Assistant. Is this correct?

[image]

BeamCtrl, 2022-03-12:
Nice, progress, I can recomend using the mqtt option to integrate with homeAssistant, gives out more entities.
[image]

however, it seems the it does not set new fanspeed values,  I’ll see if I can add some more verbosity to the log. I just see that it tries to set a new fanspeed but it fails for some reason.

overas, 2022-03-12:
How do you send the json file with mqtt publish to home assistant from the raspberry pc? My Home assistant pc have Eclipse Mosquitto broker already.

BeamCtrl, 2022-03-12:
> How do you send the json file with mqtt publish to home assistant from the raspberry pc? My Home assistant pc have Eclipse Mosquitto broker already.
I added this in my crontab file:

\* * * * * mosquitto_pub -t airiana -f Airiana/RAM/air.out

as mosquitto is running on my pi I have not defined the broker only the topic and file where the json data is.

here's my homeAssistant configuration:
[configuration.txt](https://github.com/BeamCtrl/Airiana/files/8238426/configuration.txt)
I was not allowed to attach it with .yaml extension hence the .txt

BeamCtrl, 2022-03-12:
I have a new commit up with some retry logic on TCP writes. commit: fbaf77a8

overas, 2022-03-12:
I think the commit did not help.

pi@airiana:~/Airiana $ cat ./RAM/err
Using /dev/serial0
System started	Sat Mar 12 23:37:55 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Pressure difference set to: 10%	Sat Mar 12 23:38:11 2022
('127.0.0.1', 47240):99 at	Sat Mar 12 23:38:48 2022
Changing fanspeed to:3 		Sat Mar 12 23:38:48 2022
Incorrectly set fanspeed 2 to 3 	Sat Mar 12 23:38:48 2022
('127.0.0.1', 44198):97 at	Sat Mar 12 23:39:06 2022
Changing fanspeed to:1 		Sat Mar 12 23:39:06 2022
Incorrectly set fanspeed 2 to 1 	Sat Mar 12 23:39:07 2022
('127.0.0.1', 56706):98 at	Sat Mar 12 23:39:21 2022
('127.0.0.1', 55638):8 at	Sat Mar 12 23:39:30 2022
Vent timer started at:	Sat Mar 12 23:39:30 2022
Changing fanspeed to:3 		Sat Mar 12 23:39:30 2022
Incorrectly set fanspeed 2 to 3 	Sat Mar 12 23:39:31 2022
pi@airiana:~/Airiana $

BeamCtrl, 2022-03-12:
new commit up that tries to readout the set data and if failing to set, retries. commit 8a12e7eb

overas, 2022-03-13:
Updated to the new commit, but setting the fanspeed from webinterface or ./cmd.py do not work.

```
pi@airiana:~/Airiana $ cat ./RAM/err
Using /dev/serial0
System started	Sun Mar 13 14:16:48 2022
Savecair unit set
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Pressure difference set to: 10%	Sun Mar 13 14:17:42 2022
('127.0.0.1', 49437):99 at	Sun Mar 13 14:19:23 2022
Changing fanspeed to:3 		Sun Mar 13 14:19:23 2022
Incorrectly set fanspeed 2 to 3 	Sun Mar 13 14:19:29 2022
('127.0.0.1', 43724):97 at	Sun Mar 13 14:21:14 2022
Changing fanspeed to:1 		Sun Mar 13 14:21:14 2022
Incorrectly set fanspeed 2 to 1 	Sun Mar 13 14:21:20 2022
('127.0.0.1', 59061):98 at	Sun Mar 13 14:21:25 2022
('127.0.0.1', 53462):8 at	Sun Mar 13 14:21:43 2022
Vent timer started at:	Sun Mar 13 14:21:43 2022
Changing fanspeed to:3 		Sun Mar 13 14:21:43 2022
Incorrectly set fanspeed 2 to 3 	Sun Mar 13 14:21:49 2022
('127.0.0.1', 40396):2 at	Sun Mar 13 14:22:38 2022
Changing fanspeed to:3 		Sun Mar 13 14:22:38 2022
Incorrectly set fanspeed 2 to 3 	Sun Mar 13 14:22:44 2022
('127.0.0.1', 40396):10 at	Sun Mar 13 14:22:50 2022
('127.0.0.1', 40396):1 at	Sun Mar 13 14:23:05 2022
Changing fanspeed to:1 		Sun Mar 13 14:23:08 2022
Incorrectly set fanspeed 2 to 1 	Sun Mar 13 14:23:14 2022
Dynamic fanspeed 1 without RH	Sun Mar 13 14:23:14 2022
pi@airiana:~/Airiana $ 

```

overas, 2022-03-13:
I am trying to setup mqtt publish to home assistant, but my air.out file is empty (no json data). I have restarted airiana, but the air.out do not get input data.

```
pi@airiana:~/Airiana $ cat ./RAM/air.out 
pi@airiana:~/Airiana $ 
```

[image]

BeamCtrl, 2022-03-13:
added fixes for cmd.py and air.out, seems I broke the json prints last night. commit : f1b44c77

overas, 2022-03-14:
I do not know if these info/warning/errors in /RAM/err are things you need to fix, but here they are. When I push the "[Winter/Summer Mode]" button nothing seems to happen. How can I verify the Winter mode is enabled?

```
pi@airiana:~/Airiana $ cat ./RAM/err

Using /dev/serial0
System started	Mon Mar 14 16:47:38 2022
Savecair unit set
Pressure difference set to: 10%	Mon Mar 14 16:47:53 2022
json-writer error []
Changing fanspeed to:1 		Mon Mar 14 16:57:45 2022
Dynamic fanspeed 1 with RH	Mon Mar 14 16:57:46 2022
Extract flow offset to: 0	Mon Mar 14 16:58:52 2022
Extract flow offset to: 0	Mon Mar 14 17:01:20 2022
Extract flow offset to: 0	Mon Mar 14 17:02:33 2022
Extract flow offset to: 0	Mon Mar 14 17:05:00 2022
Extract flow offset to: 0	Mon Mar 14 17:06:16 2022
Extract flow offset to: 0	Mon Mar 14 17:07:29 2022
Extract flow offset to: 0	Mon Mar 14 17:08:44 2022
Extract flow offset to: 0	Mon Mar 14 17:11:11 2022
Extract flow offset to: 0	Mon Mar 14 17:16:05 2022
Extract flow offset to: 0	Mon Mar 14 17:18:32 2022
Extract flow offset to: 0	Mon Mar 14 17:19:46 2022
Extract flow offset to: 0	Mon Mar 14 17:23:30 2022
Extract flow offset to: 0	Mon Mar 14 17:24:43 2022
Extract flow offset to: 0	Mon Mar 14 17:27:10 2022
Extract flow offset to: 0	Mon Mar 14 17:29:37 2022
Extract flow offset to: 0	Mon Mar 14 17:33:18 2022
Extract flow offset to: 0	Mon Mar 14 17:34:32 2022
Extract flow offset to: 0	Mon Mar 14 17:35:47 2022
Extract flow offset to: 0	Mon Mar 14 17:37:01 2022
Extract flow offset to: 0	Mon Mar 14 17:39:27 2022
Extract flow offset to: 0	Mon Mar 14 17:40:43 2022
Extract flow offset to: 0	Mon Mar 14 17:44:23 2022
Extract flow offset to: 0	Mon Mar 14 17:49:18 2022
Extract flow offset to: 0	Mon Mar 14 17:50:31 2022
Starting coefAI test @  17C	Mon Mar 14 17:57:50 2022
Changing fanspeed to:3 		Mon Mar 14 17:57:50 2022
Changing fanspeed to:2 		Mon Mar 14 18:07:55 2022
Dynamic fanspeed 2 with long dt from 3	Mon Mar 14 18:07:56 2022
Changing fanspeed to:1 		Mon Mar 14 18:08:01 2022
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Extract flow offset to: 0	Mon Mar 14 18:08:53 2022
Extract flow offset to: 0	Mon Mar 14 18:13:46 2022
Extract flow offset to: 0	Mon Mar 14 18:17:29 2022
Extract flow offset to: 0	Mon Mar 14 18:21:08 2022
Extract flow offset to: 0	Mon Mar 14 18:22:23 2022
Extract flow offset to: 0	Mon Mar 14 18:23:37 2022
Extract flow offset to: 0	Mon Mar 14 18:24:53 2022
Extract flow offset to: 0	Mon Mar 14 18:26:07 2022
Extract flow offset to: 0	Mon Mar 14 18:27:21 2022
Extract flow offset to: 0	Mon Mar 14 18:28:35 2022
unable to ping, network error	Mon Mar 14 18:32:55 2022
Extract flow offset to: 0	Mon Mar 14 18:34:45 2022
Extract flow offset to: 0	Mon Mar 14 18:37:11 2022
Extract flow offset to: 0	Mon Mar 14 18:43:18 2022
Extract flow offset to: 0	Mon Mar 14 18:44:33 2022
Extract flow offset to: 0	Mon Mar 14 18:46:59 2022
Extract flow offset to: 0	Mon Mar 14 18:48:14 2022
Extract flow offset to: 0	Mon Mar 14 18:51:55 2022
Extract flow offset to: 0	Mon Mar 14 18:53:10 2022
Extract flow offset to: 0	Mon Mar 14 18:56:50 2022
Extract flow offset to: 0	Mon Mar 14 18:59:18 2022
Extract flow offset to: 0	Mon Mar 14 19:00:31 2022
Extract flow offset to: 0	Mon Mar 14 19:02:59 2022
Extract flow offset to: 0	Mon Mar 14 19:05:26 2022
Extract flow offset to: 0	Mon Mar 14 19:07:52 2022
Extract flow offset to: 0	Mon Mar 14 19:10:21 2022
Extract flow offset to: 0	Mon Mar 14 19:11:35 2022
Extract flow offset to: 0	Mon Mar 14 19:12:48 2022
Extract flow offset to: 0	Mon Mar 14 19:14:03 2022
Extract flow offset to: 0	Mon Mar 14 19:16:29 2022
Extract flow offset to: 0	Mon Mar 14 19:20:11 2022
Extract flow offset to: 0	Mon Mar 14 19:21:24 2022
Extract flow offset to: 0	Mon Mar 14 19:23:52 2022
Extract flow offset to: 0	Mon Mar 14 19:25:05 2022
Extract flow offset to: 0	Mon Mar 14 19:26:23 2022
Extract flow offset to: 0	Mon Mar 14 19:32:29 2022
Extract flow offset to: 0	Mon Mar 14 19:34:56 2022
Extract flow offset to: 0	Mon Mar 14 19:36:11 2022
Extract flow offset to: 0	Mon Mar 14 19:38:38 2022
Extract flow offset to: 0	Mon Mar 14 19:42:17 2022
Extract flow offset to: 0	Mon Mar 14 19:43:33 2022
('127.0.0.1', 59152):0 at	Mon Mar 14 19:44:54 2022
cycle exchanger to: None	Mon Mar 14 19:44:54 2022
('127.0.0.1', 35745):1 at	Mon Mar 14 19:45:19 2022
('127.0.0.1', 58293):1 at	Mon Mar 14 19:45:32 2022
cycle exchanger to: 5	Mon Mar 14 19:45:36 2022
1.Exchange set to 5. extract is less than target-1C 	Mon Mar 14 19:45:36 2022
Extract flow offset to: 0	Mon Mar 14 19:47:14 2022
Extract flow offset to: 0	Mon Mar 14 19:49:41 2022
Extract flow offset to: 0	Mon Mar 14 19:50:55 2022
Extract flow offset to: 0	Mon Mar 14 19:52:10 2022
Extract flow offset to: 0	Mon Mar 14 19:54:37 2022
Extract flow offset to: 0	Mon Mar 14 19:55:50 2022
Extract flow offset to: 0	Mon Mar 14 19:57:05 2022
Extract flow offset to: 0	Mon Mar 14 19:58:19 2022
Extract flow offset to: 0	Mon Mar 14 19:59:32 2022
unable to ping, network error	Mon Mar 14 20:00:33 2022
pi@airiana:~/Airiana $ 
```

BeamCtrl, 2022-03-14:
When in summer mode the exchanger will stop after a little while. You can see exchanger mode flip from 5 to 0. 

Something is up with offset to zero should only print once.
//

overas, 2022-04-06:
Updated to 52e34ed and everything works fine. I get a error about "json-writer error []" when the airiana service start up, but forecast.json and air.out are updated with data so I think is only a startup error.

I have a question about forced ventilation. I tried to use the service "RESTful Command: force_vent" in HA to force more air when fire up the fireplace. How can I controll the time the fan is on and how can I change back to auto? Is it better to use the service "RESTful Command: vent_high" with a time delay and then use "RESTful Command: vent_low" for stopping the fan.

[image]

`
pi@airiana:~/Airiana/RAM $ cat err

Using /dev/serial0
System started  Wed Apr  6 13:40:15 2022
Savecair unit set
Pressure difference set to: 10% Wed Apr  6 13:40:28 2022
json-writer error []
('127.0.0.1', 50634):99 at      Wed Apr  6 13:44:09 2022
Changing fanspeed to:3          Wed Apr  6 13:44:09 2022
('127.0.0.1', 40824):1 at       Wed Apr  6 13:44:45 2022
('127.0.0.1', 59135):1 at       Wed Apr  6 13:44:58 2022
Changing fanspeed to:2          Wed Apr  6 13:45:01 2022
Dynamic fanspeed 2 with long dt from 3  Wed Apr  6 13:45:02 2022
pi@airiana:~/Airiana/RAM $
`

overas, 2022-04-06:
Has something changed with the temperature picture in the webpage? The picture has on the left side black bars which is not pretty.

[image]

BeamCtrl, 2022-04-06:
Have you updated? The black bars looks like the is temp values in the thousands, the bars being temp levels stacked on top of each other, the graph looks strange in general.  Pull latest and see if there is a change

overas, 2022-04-07:
I started with a fresh install like you adviced here: https://github.com/BeamCtrl/Airiana/issues/22#issuecomment-1064521629.
But the airiana service do not start up and the log says the user pi do not have permission to create data.log, forecast.txt, err. https://gist.github.com/overas/7c9b49e999d9ec7efe4618b2c77c6f4e.
I then replaced the user pi in /etc/systemd/system/airiana.service with root and the service start up. The files is also created. 

I do not see any new button for fireplace in the webpage. Should this work as a rest_command also in HA?

[image]

BeamCtrl, 2022-04-07:
Its the fire start 10 min one, it is new . Rest command is correct, this one is nr ”9”.

Hämta Outlook för iOS<https://aka.ms/o0ukef>

[image]<https://user-images.githubusercontent.com/37910044/162258028-d7ed3b99-5429-4c0a-99d8-27d231434ee5.png>

overas, 2022-04-07:
I saw the new icon now after I refresh the webpage. The fire start button works as expected.

Fire start at:  Thu Apr  7 21:17:28 2022
Pressure difference set to: -10%        Thu Apr  7 21:17:28 2022
Pressure difference set to: 10% Thu Apr  7 21:26:15 2022
Changing fanspeed to:1          Thu Apr  7 21:27:45 2022
Changing fanspeed to:3          Thu Apr  7 21:27:47 2022
cycle exchanger to: 0   Thu Apr  7 21:28:59 2022
Exchange set to 0 supply>target         Thu Apr  7 21:29:00 2022
cycle exchanger to: 5   Thu Apr  7 21:29:00 2022
1.Exchange set to 5. extract is less than target-1C     Thu Apr  7 21:29:00 2022

> Has something changed with the temperature picture in the webpage?

After I did a fresh install of Airiana the temperature picture is now correct.

BeamCtrl, 2022-04-07:
Great to hear things are working, yes I had to do some changes to the graph, not sure why but the border was hiding the temperature legend. I had not changed anything so I suspect some dependencies in matplotlib to be the cause.

overas, 2022-04-11:
> After I did a fresh install of Airiana the temperature picture is now correct.
The problem with the temperature picture has come back. The picture has also gray horizontal stripes in the background.

[image]

BeamCtrl, 2022-04-12:
I think the problem is with the data. I looks as you are getting data in at around a few hundred Celcius. Can you attach or look in the data file "data.log" for entries around this date use [this converter](https://www.epochconverter.com/) on the timestamps to find where this happened.

overas, 2022-04-15:
Sorry for late reply. I attach the "data.log" file. I think it is easier if you look at it. The graph is normal now, but I hope you can find it out.
[data.log](https://github.com/BeamCtrl/Airiana/files/8497859/data.log)

BeamCtrl, 2022-04-17:
had a look thru of your data file, it seems that the systemair unit is reporting temperatures in excess of 4000C at a few points. The grey lines are the 1C step lines that get pushed together. I'm not sure why you are getting "hell-level" temperatures sometimes,

overas, 2022-04-18:
Thanks for your reply. Today in the nighttime I had the same insane temperatures for the inlet temperature. I am going to try the new fix you made today. 

[image]

overas, 2022-04-22:
Now I think I have a problem with the humidity. It seems the calculated outside humidity is out of range. After a shower the fan do not go from low to high anymore.

[image]

[data 22-04-2022.log](https://github.com/BeamCtrl/Airiana/files/8543650/data.22-04-2022.log)

BeamCtrl, 2022-04-22:
something is very strange with your sensor data, the indoor temperature seems to have flat lined at 0.0C, this causes the humidity calculations to err. however the shower mode should still kick in as this is only dependent on the humidity sensor and that data looks intact. Have you tried rebooting the whole systemair unit?

overas, 2022-04-23:
Thanks again. I rebooted the VTR500  and the IAM and it seems this has helped. The graphs look correct now with outside temperature 8.9 degree and a humidity 30 % at 06:57.

[image]
[data 23-04-2022.log](https://github.com/BeamCtrl/Airiana/files/8546233/data.23-04-2022.log)

[image]

overas, 2022-04-25:
> however the shower mode should still kick in as this is only dependent on the humidity sensor and that data looks intact

The "Shower detections" is not working. After a shower the fans do not go from Low to High. I can not remember which commit made the problem, but it has worked. The reboot of the IAM and the fan has not helped either. 

The calculated outside humidity is way off in the picture. It seems the data is to high so the graph starts to go above 100 %.

I have also in the err log a few times:
`Write error, no tries left on register:1403 None
:30:04 2022

[data.log](https://github.com/BeamCtrl/Airiana/files/8557244/data.log)
[request.log](https://github.com/BeamCtrl/Airiana/files/8557245/request.log)
[err.txt](https://github.com/BeamCtrl/Airiana/files/8557249/err.txt)

`

BeamCtrl, 2022-04-25:
Not sure about the cause of this but it seems the extract temp is 0 again. This will explain RH above 100% if temp outside is higher than 0. The log ends on the 22:nd and it looks strange at the end with regards to the extract temp. Maybe you can run ./backup.py to get the last 25hrs from RAM into the data.log file before posting to have it include the latest data. Airiana does this intermittently to save on SD-card writes and it is kind of wierd I agree.

overas, 2022-04-30:
Here are the data.log after running ./backup.py and the err file.  I get a lot of:

`unable to ping, network error   Fri Apr 29 11:11:08 2022

in the err file. Can this explain the missing relative humidity?
`
[Airiana-data.log](https://github.com/BeamCtrl/Airiana/files/8596472/Airiana-data.log)
[RAM-data.log](https://github.com/BeamCtrl/Airiana/files/8596473/RAM-data.log)
[err.txt](https://github.com/BeamCtrl/Airiana/files/8596474/err.txt)

overas, 2022-04-30:
From the HA history graph it looks like the relative humidity is above 100 % two times.

[image]

BeamCtrl, 2022-04-30:
Yes this happens when your unit starts to report zero values on extract temperature, very unclear why it shifts sometimes.

overas, 2022-04-30:
After a restart of the airiana service the values is back again to normal.

overas, 2022-05-02:
I am trying to do a fresh install of Airiana following your advice here: https://github.com/BeamCtrl/Airiana/issues/22#issuecomment-1064521629

I know get an error with install.py

```
pi@airiana:~/Airiana $ sudo python3 install.py clean
Traceback (most recent call last):
  File "/home/pi/Airiana/install.py", line 19, in <module>
    user_id = pwd.pwuid()
AttributeError: module 'pwd' has no attribute 'pwuid'
pi@airiana:~/Airiana $
```

BeamCtrl, 2022-05-03:
fix to pwuid is up,
 just to explain some of your questions above, unable to ping means that the unit is not able to contact the internet and do a DNS lookup, "no internet connection". 
 
The reason that the RH is above 100 is due to the fact that when it is receiving 0C as indoor temp, it calculates what the outside max  possible RH level is expressed as a relation to the indoor temp. If the outdoor is higher than 0C this will result in a above 100% RH as the 0C would generate condensation if outside air were to move inside.

overas, 2022-05-06:
> fix to pwuid is up,

The fix do not work for me. 

```
pi@airiana:~/Airiana $ whoami
pi
```

```
pi@airiana:~/Airiana $ sudo python3 install.py clean
Traceback (most recent call last):
  File "/home/pi/Airiana/install.py", line 19, in <module>
    user_id = pwd.getpwuid()
TypeError: pwd.getpwuid() takes exactly one argument (0 given)
```

BeamCtrl, 2022-05-07:
tnx for reporting now fixed properly, was a mistake on my part.

overas, 2022-05-08:
> The reason that the RH is above 100 is due to the fact that when it is receiving 0C as indoor temp, it calculates what the outside max possible RH level is expressed as a relation to the indoor [...]

Is this something you can fix in Airiana? Or is it something wrong with my fan or IAM?
Today my wife and I have showered twice. Airiana do not sense the high moisture in the bathroom and the fan does not change to high to lower the humidity.

[image]

overas, 2022-05-08:
I think Airiana configured my VTR500 fan to have airflow "Low" and "Temperature" 24 degree now. I want to have airflow Normal. How can I configure Airiana to use Normal as default?
I have tried Fanspeed Normal but it do not work.

BeamCtrl, 2022-05-09:
quite the challange, I'm reluctant to blame hardware, I have however been unable to find a reasonable cause within the airiana code to

BeamCtrl, 2022-05-09:
Noted that since we are getting the proper RH values I may be able to work around the shower issue but it will take a little time to figure out the actual implications of changing the logic. It may impact other things.

overas, 2022-05-20:
Sorry for late reply. I have been testing Home Assistant modbus sensor integration for trying to find out why the unit starts to report zero values from the extract temperature. This do not happen as a sensor in HA. 

But I had problems with IAM reporting in the HA log it was busy. I have tried changing the ethernet cabel and not using a switch and using av delay in the config, but it did not help. After a while with troubleshooting I found out from the systemair documentation https://shop.systemair.com/upload/assets/SAVE_MODBUS_VARIABLE_LIST_20190116__REV__29_.PDF, I must have in the config the signal type used.

[image]

From the HA modbus doc https://www.home-assistant.io/integrations/modbus/, the Unsigned integers use data_type: uint16 and Signed integers use data_type: int16. 

      - name: "VTR500 Extract Air Temperature"
        unit_of_measurement: °C
        device_class: temperature
        slave: 1
        address: 12543
        scale: 0.1
        precision: 1
        data_type: uint16

I tried to find in the airiana-core.py if unsigned integers is specified for address: 12543, but my knowledge in Python is not so good. The pymodbustcp libary have this documentation about this https://pymodbustcp.readthedocs.io/en/latest/quickstart/index.html#utils-module-modbus-data-mangling

So here comes the question. Could it be that pymodbustcp is reading wrong from address: 12543 and give zero or something else which airiana thinks is temperature 0 degree?  Look here: https://github.com/sourceperl/pyModbusTCP/issues/22

BeamCtrl, 2022-05-20:
Hi, data type is correct as it works most of the time, any issues with sign bit will be seen if extract temp would be lower than 0 and then as a twos complement. However I had some ideas reading thru the modbusTCP documentation on what would happen if there is a loss of connection mid the communication cycle. Did not howver become any smarter but added Auto close festure on the connection to reconnect after each request along with some more error prints.

overas, 2022-05-21:
It is something wrong with my IAM I think. I thought I had solved the SlaveBusy error, but it comes back again.

[image]

Do you use a IAM to communicate?

[12 more replies]

closed 2022-10-06 by BeamCtrl