BeamCtrl/Airiana #23: Exhaust temp missing with VR400

DrSammie, 2022-01-30:
**Describe the bug**
Exhaust temp reading is missing with Systemair VR400

**To Reproduce**
Steps to reproduce the behavior:
1. Systemair VR 400.
2. Fresh install of Raspbian GNU/Linux 10 (buster), RS 485 via "/dev/ttyUSB0"
3. Installed Airiana
4. No exhaust temp in web interface

**Expected behavior**
To se the exhaust temp displayed and be used.

**Additional context**
I looked back at previous versions of the code of ariana-core.py and found a change in the code that could be the cause of this. Commit 7067b1936c42c15d7ac69629e6bea8749146dcd5, 7 nov 2021 and the changes added to line 816 to 821
I tried to change the code to:
" def update_xchanger(self):
                if len(self.inlet):
                        self.inlet_ave = numpy.average(self.inlet)
                        self.supply_ave = numpy.average(self.supply)
                        self.extract_ave = numpy.average(self.extract)
                        if self.system_name == "VR400":
                                self.exhaust_ave = numpy.average(self.exhaust)
                else:
                        self.inlet_ave = self.inlet[0]
                        self.supply_ave = self.supply[0]
                        self.extract_ave = self.extract[0]
                        #self.exhaust_ave =  self.exhaust[0]
                        if self.system_name == "VR400":
                                self.exhaust_ave = self.exhaust[0]"

which seems to work, but there is a 2-3 min hang up after the first read but then it seems to be good.

Thank you for a very nice program!
And take my comments with the cosideration that my programming experience come from now 20 years old school courses with C++ and Mircosoft basics about 20 years ago. So my finding and quick fix may be completly wrong and break other stuff ;)

best regards

Samuel

BeamCtrl, 2022-01-30:
Good catch and thank you for debugging. I have since a while back replaced my VR400 with a VTR300 and these kinds of bugs are hard to find without a physical unit. I added your changes and pushed it to the master branch.

closed 2022-01-30 by BeamCtrl

DrSammie, 2022-02-01:
I´m glad that it helped. Thanks again for a great piece of software.

commit aeced203 (2022-01-30): closes #23 moved average calculation on exhaust temperature  from first pass back to recurrent reading. also added VR400 case on first pass. cred to DrSammie for the ticket.
commit 6952f68b (2022-01-30): closes #23 incorrect handling of the VR400 exhaust initial values, cred. MrSammie