BeamCtrl/Airiana #24: json-writer error []

einarjh, 2022-10-09:
**Describe the bug**
The file air.out remains empty and the RAM/err file is filled with the error message "json-writer error []"

The rest seems to be working.

**To Reproduce**
Steps to reproduce the behavior:
I started with a blank Raspberry Pi 4 with the latest raspbian lite, and ran the install command.

**Expected behavior**
Sensor data is available for publishing to mqtt.

**Screenshots**
This is a paste from the airiana web frontend: 
```
VTR300 Sun Oct 9 13:21:43 2022 status: 795(3410)0.23 Vers. 10.3 ***
Inlet: 11.32C    Supply: 20.00C  d_in : 8.69C
Extract: 21.32C  Exhaust: 13.45C  d_out: 7.87C
Extract dT/dt: 0.001degC/min dT/dt: 0.371degC/hr

Exchanger Setting: 5 State: Normal, Rotor Active: Yes
HeatExchange supply 377.7W
HeatExchange extract 377.9W
Pressure limit: 11.82C

Electric power:77.0W COP:4.9
Energy Loss: 160.9W
Loss total: 0.039kWh Average:174.8W
Cooling total: 0.02kWh
Heat gain total: 0.0kWh
Unit electric total: 0.014kWh
Supply:36 l/s,3060rpm  Extract:40 l/s,3300rpm
Temperature Efficiency: 84.35%
Filter has been installed for 2 days ,99.5% remaining.

Ambient Pressure:996.3hPa
Weather forecast: 12.1C 100.0% cloud cover RH:70.8%

Pressure change inhibited (18min 56s)
Exchanger mode change inhibited (46min 45s)
```

**Desktop (please complete the following information):**
 - OS: Raspbian GNU/Linux 11 (bullseye)
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

BeamCtrl, 2022-10-09:
Hello, I can understand that the json-writer error could be a bit more informative. Did a push just now to include a lite more meat to that bone. However this occurs when the Airiana tries to put the humidity data to the json-file. If humidity data is missing for some reason the call with this strangely worded error. Do you know if your units humidity sensor is malfunctioning or is incorrectly configured? I will do an update shortly to address the need for humidity data, might take me a few minutes.

BeamCtrl, 2022-10-09:
ok, I just pushed  9616a5c9 give it a try by doing a `git pull && ./restart`

einarjh, 2022-10-09:
I'm not aware of any malfunctions or misconfigurations, though how could I tell? The ventilation system has seemed to be working fine, this is the first time I'm hooking up to it via modbus.

The system detected my shower yesterday morning, though, and it produces humidity graphs: [example](https://eina.rjh.im/g/WXA4GS.png)

I updated to 9616a5c and still getting the same error. note also a numpy error that I've seen before, but it had disappeared from my log on my initial bug report: 

```
Using /dev/ttyUSB0
Holdoff time in effect, will re-ping in one hour.
System started	Sun Oct  9 18:44:12 2022
Holdoff time in effect, will re-ping in one hour.
Pressure difference set to: 10%	Sun Oct  9 18:45:08 2022
json-writer error []
/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py:3372: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/usr/lib/python3/dist-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
json-writer error []
json-writer error []
json-writer error []
....
```

BeamCtrl, 2022-10-09:
did it really update the json-writer error should be more informative in the new version.
[image]
try to pull the latest and restart, see if you get the updated error msg

einarjh, 2022-10-09:
Seems I hadn't gotten the latest update after all (I used the update.py script) - now it seems to work, no more error messages! But now, of course, I'm missing out on "measured humidity" and "refererence humidity" values:
```
{
  "extract": 21.91044444444444,
  "coolingMode": "false",
  "supply": 19.834222222222227,
  "sf": 91,
  "ef": 100,
  "exhaust": 12.110444444444605,
  "autoON": "true",
  "shower": "false",
  "rotorSpeed": 100,
  "sfRPM": 6300,
  "energyXfer": 584.4749471335036,
  "efficiency": 83.70358142303336,
  "efRPM": 6180,
  "name": "VTR300",
  "filterPercentRemaining": 99.5,
  "pressure": 994.3,
  "filterInstalledDays": 2,
  "rotorActive": "yes",
  "elecHeater": 0,
  "inlet": 9.169241111111111,
  "electricPower": 446.8043855191196,
  "electricPowerTotal": 236.06
}
```
I'm curious; why does it show up in the graphs?

BeamCtrl, 2022-10-09:
Quick answer; it is not there because you probably do not have an installed/configured humidity sensor; the humidity level you see in the diagram is a calculated one. I’ll fiddle a bit to get you the data you want in the json-file but you may want to check your units spec and or installation, I have a vtr300 aswell and it has an rh-sensor.

BeamCtrl, 2022-10-09:
a fix is up, now added CalculatedHumidity to the json output file

einarjh, 2022-10-09:
That's awesome, thanks!

I had a look at the installation and service manual for the VTR 300, and the diagram for the print card clearly indicates a terminal header for a humidity sensor. This headed is missing from my print card. I'm at a loss. This sensor, is it meant to be within the unit itself, or is it an external sensor somewhere in the ducts?

einarjh, 2022-10-09:
I'm even more confused, now I'm seeing all three measurements:
```
{
  "extract": 21.778700542489318,
  "coolingMode": "false",
  "supply": 20.179055555555554,
  "sf": 36,
  "ef": 40,
  "exhaust": 11.938691467216982,
  "autoON": "true",
  "shower": "false",
  "rotorSpeed": 100,
  "sfRPM": 3060,
  "energyXfer": 195.9557019197502,
  "efficiency": 87.14653664030423,
  "efRPM": 3240,
  "name": "VTR300",
  "filterPercentRemaining": 99.5,
  "pressure": 994.3,
  "filterInstalledDays": 2,
  "rotorActive": "yes",
  "elecHeater": 0,
  "inlet": 9.334068333333336,
  "electricPower": 75.00377635074058,
  "electricPowerTotal": 13.51,
  "calculatedHumidity": 43.8,
  "referenceHumidity": 39.91472108522595,
  "measuredHumidity": 43.79398610486934
}
```

einarjh, 2022-10-09:
Should "rotorActive" be a boolean, and a binary sensor? Or will it sometimes say "maybe"? :grin:

BeamCtrl, 2022-10-12:
closing after running live for 48hrs

closed 2022-10-12 by BeamCtrl