BeamCtrl/Airiana #26: Airiana and polar nights

jdybvik, 2022-12-05:
**Describe the bug**
When using Airiana at latlong position above the arctic circle (in polar nights), a fatal error occur, causing forecast not to be loaded, and eventually total crash of communication with systemair. Error log: 

et(body, start, use_center, +1, True)
  File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 497, in _find_rise_or_set
    % (body.name, self.date))
ephem.NeverUpError: 'Sun' is below the horizon at 2022/12/4 10:43:56
Forcast does not return proper data. Sat Dec  3 21:22:39 2022
Traceback (most recent call last):
  File "/home/pi/Airiana/airiana-core.py", line 2041, in <module>
    device.get_local()
  File "/home/pi/Airiana/airiana-core.py", line 1889, in get_local
    if time.localtime().tm_hour == sun and time.localtime().tm_min < 5 or self.prev_static_temp == 8:
UnboundLocalError: local variable 'sun' referenced before assignment

Using /dev/ttyUSB0
System started	Sat Dec  3 21:22:40 2022
Traceback (most recent call last):
  File "./forcast2.0.py", line 65, in <module>
    rise, setting = get_sun(latlong["lat"], latlong["long"])
  File "./forcast2.0.py", line 22, in get_sun
    return ephem.localtime(o.next_rising(s)), ephem.localtime(o.next_setting(s))
  File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 439, in next_rising
    return self._find_rise_or_set(body, start, use_center, +1, True)
  File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 497, in _find_rise_or_set
    % (body.name, self.date))
ephem.NeverUpError: 'Sun' is below the horizon at 2022/12/4 10:43:56
Traceback (most recent call last):
  File "./forcast2.0.py", line 65, in <module>
    rise, setting = get_sun(latlong["lat"], latlong["long"])
  File "./forcast2.0.py", line 22, in get_sun
    return ephem.localtime(o.next_rising(s)), ephem.localtime(o.next_setting(s))
  File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 439, in next_rising
    return self._find_rise_or_set(body, start, use_center, +1, True)
  File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 497, in _find_rise_or_set
    % (body.name, self.date))
ephem.NeverUpError: 'Sun' is below the horizon at 2022/12/4 10:43:56
Forcast does not return proper data. Sat Dec  3 21:22:46 2022
Traceback (most recent call last):
  File "/home/pi/Airiana/airiana-core.py", line 2041, in <module>
    device.get_local()
  File "/home/pi/Airiana/airiana-core.py", line 1889, in get_local
    if time.localtime().tm_hour == sun and time.localtime().tm_min < 5 or self.prev_static_temp == 8:
UnboundLocalError: local variable 'sun' referenced before assignment

**To Reproduce**
Using current Airiana configured with latlong position above arctic circle in polar nights. 

**Expected behavior**
Airiana function perfectly with latlong position below arctic circle. 

**Additional context**
Temporarily fixed by changing latlong.json position.

BeamCtrl, 2022-12-10:
Thank you for a neat bug find. I did not consider the earth axis tilt in the initial designs ;). added a fix that uses a standard day of 0600-1900 when the polar night / midnight sun is doing its thing, update avail from commit `cc086310`.

//Daniel

BeamCtrl, 2022-12-25:
idle for two weeks will consider it closed.

closed 2022-12-25 by BeamCtrl