BeamCtrl/Airiana #27: Water condensation on windows inside with VR400

DrSammie, 2022-12-22:
Hi,
I have used Ariana for about a year and it has been working well. I really appreciate how well the program works.

I have had low fan speeds most of the time, supply 36 l/s and extract 40 l/s. Noticed now during the last weeks that with the lowest temperatures outside (-15 to -23C) that there was a lot of  water condensations on the windows inside. Indoor humidity during this time was RH35-40%, not measured in the VR400.
I´ve been looking at the problem and I think the cause is a combination of the following:

1. We have reduced the indoor heating to 19-20C from previous target of 22C. Which reduces the fanspeeds to try to increase 		 	temp to target of 22 in Ariana.
2. A relative higher humidity inside compare to the outdoor
3. The default lowest fanspeed is to low for our house. Looked at the OVK-papers from when the house was built 2014 and the flow there should be 16h of supply 80 l/s and extract 90 l/s.	

I tried to edit airiana-core.py with the following changes which seemed to solve the problems I experienced.
Line 477	self.target = 19 #was 22
Line 550	req.write_register(102, 60) #was 40
Line 551	req.write_register(103, 80) #was 60
Line 552	req.write_register(104, 90) #was 60

With regards
Samuel

BeamCtrl, 2022-12-25:
yes you are correct in you analysis, increased flow should help with condensation when it is cold. If at 23C indoor any surface below 8-6C if at 35-40%RH, condensation will occur.

The fix however is not optimal for your installation. addr:103 is supply fan flow for medium speed and addr:104 is for extract fan flow medium speed,  addr:102 is supply fanspeed for low flow, To help with the issue you should only need to change the value of address 102 to say 40 for increased low flow or what flow you need for the low level, this will still give the airiana unit the ability to regulate depending on indoor conditions, setting target 19 will only push it into higher flows always. for the medium flow lvls it is recommended to keep them identical as the ratio between them is what controls the over/under pressure in the building, 

There is always a trade off between efficiency and airflow, I created Airiana fot this reason, the shower mode is intended to evacuate alot more of the indoor moisture as it is generated witch in my case warrants a lower contiguous flow. With this said it is important to adapt the settings to what ever circumstance is valid for each installation. I will be adding flow config options to the Web gui in the future. Untill then go with the manual tweak opiton. =).
 

best rgrds //Daniel

DrSammie, 2022-12-27:
Hi, thank you for your comment. 
Have tried to change addr:102 and use equal values on addr:103 and addr:104. And it seams to work better for me.

Wouldn´t setting target temp to the same temperature as the target for the heating systems in the house make sense in my case?

I think that Airiana handels the trade of between efficiency and airflow very well.  And before I started using Airiana the air was to dry indoor during the winter so it is much better now, with the exception with the humidity on windows during the lowest outdoor temperatures.
Unfortunately the shower mode does not work for me. Bu I´m pretty sure the problem is that there is to many air outlets in our house so that the temperature increase is to small to active shower mode. And since it’s the VR400 there is no humidity sensor either. Thinking about trying to add separate esphome humidity sensors to monitor the inlet and the extract, just to be able to monitor it.

I´ll continue to with som manual tweaking ;)

Thanks again for a good program.

best regards
Samuel

BeamCtrl, 2022-12-27:
The target temperature is only the target for when the system considers to lower flows it does not try to enable the electric heater to add energy to the air flow. As such you can consider the target more as a offset for when increased air flow is required. The target changes depending on what the forecast says about the coming days and depending on the inlet temperature. It allows for a higher indoor temp when in winter time compared to summer for instance. The risk you may encounter by setting it low is that the system will try to disable the heat recuperation if the extract temp ends up a little higher than expected for some reason (cooking, guests, sauna) and I'd say its better to keep the heat we have rather than evacuating it, at least until a level where it is out of comfort zone. anyway only my recommendation but that is the design idea anyway. 

If you want the shower mode to work at a lower setting you will need to tweak the shower detection limit, 

1196 ` lim = 0.07`

to some lower value. It is the short term derivative of the exhaust temperature, hence if the short term temperature gradient is greater than 0.07C/min shower mode on. You can readout the value in the web view as dT/dt degC/min. to see how much of a response you get when someone is in the shower.

If you put this to low and it starts sensing false positives eg. the gradients drops within 2minutes of engaging it will automatically add to the limit to make it less sensitive, so its fairly benign to fiddle with and "should" auto tune itself.

//Daniel

BeamCtrl, 2023-01-09:
Closed as of inactivity

closed 2023-01-10 by BeamCtrl