BeamCtrl/Airiana #28: Since last version (python3 version) temperatures to HomeAssistant does no longer work with homeAss option.

DrSammie, 2022-12-22:
Hi,
Thank you for a great program!¨

Just a small bug I found. I have the VR400. After the last pushed upgrade (the python3 version) temperatures is not reported to home assistant with the home.Ass feature.
Had it working before the upgrade and it partially works now since ftx_Efficiency,  humidity and fanspeed works and gives values. But  ftx_Outside and  ftx_Indoor gives no values. 

Looked at the data.log and the entries looks like this

1671616751.1622722:0:19.91:0:35.33:3.24:4.34:15.2:35.62:0:-495.8:0:819.9371660149282:825.7139182024873:-5.776752187559168

 If I read and understand the code correct it seems that device.extract_ave and device.inlet_ave does not have values. I tired to find the solution myself but could not figure it out at the basic level I´m at.

I understand that the homeAss solution, is the older solution and that the MQTT option is a better way forward. However I am not familiar on how to publish files as mqtt payloads so I´m stuck there. A more description step by step on how to get a air.out published to HomeAssistant would be greatly appreciated

With regards
Samuel

BeamCtrl, 2022-12-25:
Thank you for the kind words, and a merry christmas. I updated the Readme with install instructions for mosquitto light weight MQTT broker, It is what I use and I have not had any major issues with it, It will also give you a whole lot more input to homeAssistant. 

regarding the data bug I am not able to reproduce it on my end, I have populated averages both for indoor and outside. I recommend you transition to MQTT.

I added a file homeAssistantConfig to the repo with examples of how I setup the sensors for my HomeAss integration.

pls give me a heads up on your progress, if you manage to get it running I'll close the ticket,

DrSammie, 2022-12-25:
A merry chistmas to you to!

I got the MQTT broker working. Had to do "sudo apt get install -y mosquitto-clients" since mosquitto_pub was not included with the mosquitto package in the debian distribution I use. Change the contab entry a little bit to get it working for me since I run Home Assistant on a different raspberry pi. 
"* *  * * *  mosquitto_pub -h 192.168.1.167 -p 1883 -t airiana -f /home/pi/Airiana/RAM/air.out
@reboot mosquitto & 2&>/dev/null"

The MQTT approach is much better and as you say gives a whole lot more input. Will use this instead of the homeAss approach. 
Thank you for the help, and a big thumbs up for the "Fire start 10 min" feature as it works great!

BeamCtrl, 2022-12-27:
great will add the clients to the instructions just to be extra verbose.

closed 2022-12-27 by BeamCtrl