BeamCtrl/Airiana #30: Installation oin a raspberry pi 4 with HA os

Ontrackx, 2023-01-09:
Hello, i have a rasberry pi4 running HA os

For me a novice, it is not clear how to install this. '
Also a prewiev ow the web gui would have been nice :)

Anyone that can point me in the right direction?
Rpi4 wit ha os
HF2221 Modbus over tcp/ip i also have the IAM module. IAM is only for the app purpose
Modbus HF2221 
192.168.86.41
Port 502
Buffer size 512
Keep alives 60
Timout 0

Protocol settings
Max accsept 32

Security disabeled
Route UART1

Serial port
Baud 38400
81N

Buffer 512
Gap time 50
Flow control half duplex
Cli disabled
Protocol setting modbus

I actually have no clue if this is the correct settings. But it works with another setup i have

BeamCtrl, 2023-01-10:
Hello, If you have the IAM just follow the IAM instructions for getting it to work. I am not sure about what is different if any from RpiOS to HaOS. Anyway you should be able to use Airiana with only the IAM configuration present.

Good idea with some screenshosts, Ill get on it.

Ontrackx, 2023-01-10:
Thank you for the answer, yes i understand.
But it is the part of cloning the git via ssh commandline in Homeassistant on the rpi i dont quite understand 
Commands to use, phyton 3 is installed but i get the error
Solved it had to go into  cd /config after ssh\ing in

1. git clone https://github.com/BeamCtrl/Airiana.git       ok
  ➜  ~ cd /config
➜  /config git clone https://github.com/BeamCtrl/Airiana.git
Cloning into 'Airiana'...
remote: Enumerating objects: 3774, done.
remote: Counting objects: 100% (856/856), done.
remote: Compressing objects: 100% (274/274), done.
remote: Total 3774 (delta 597), reused 835 (delta 581), pack-reused 2918
Receiving objects: 100% (3774/3774), 752.98 KiB | 3.45 MiB/s, done.
Resolving deltas: 100% (2586/2586), done.
➜  /config python3 ./install.py
python3: can't open file '/config/./install.py': [Errno 2] No such file or directory
➜  /config dir

What am i doing wrong here?

BeamCtrl, 2023-01-10:
try, 
```
git clone https://github.com/BeamCtrl/Airiana.git 
cd Airiana
python3 install.py
```
not sure what you are trying to reference, note that  "/" is top level root dir,  "~/"  is your user home dir, and "./" is current dir.

Ontrackx, 2023-01-10:
It is so weird i can see the folder if i use ls
and i can see the folder with content but cant navigate to it
Airiana                   home-assistant.log.fault  scenes.yaml
VSR                       home-assistant_v2.db      scripts
automations.yaml          home-assistant_v2.db-shm  scripts.yaml
blueprints                home-assistant_v2.db-wal  secrets.yaml
configuration.yaml        image                     sensor.yaml
custom_components         input_boolean             switches
customize                 input_datetime            traccar.xml
deps                      input_number              tts
esphome                   input_select              www
global                    known_devices.yaml        zigbee.db
groups.yaml               nest                      zigbee.db-shm
home-assistant.log        node-red                  zigbee.db-wal
home-assistant.log.1      rtsp2webrtc_v5_aarch64
➜  /config
➜  /config

it is zsh i do not know if it differs from normal ssh

Ontrackx, 2023-01-10:
Aw crap after using cd /config  
just write the folder i wanted to go into

Next problem i think i have cloned it to the wrong folder subfolder sould it be cloned to the root system ?
`  Airiana git:(master) python3 ./install.py
Installing the AirianaCores
setup autostart for airiana.service
Traceback (most recent call last):
  File "/config/Airiana/./install.py", line 124, in <module>
    setUart()
  File "/config/Airiana/./install.py", line 23, in setUart
    boot_file = open("/boot/config.txt", "r+")
FileNotFoundError: [Errno 2] No such file or directory: '/boot/config.txt'
cp: can't create '/etc/systemd/system/': No such file or directory
sudo: systemctl: command not found
setup autostart for controller.service
cp: can't create '/etc/systemd/system/': No such file or directory
sudo: systemctl: command not found
Traceback (most recent call last):
  File "/config/Airiana/./install.py", line 159, in <module>
    if boot_cmd not in open("/boot/cmdline.txt").read():
FileNotFoundError: [Errno 2] No such file or directory: '/boot/cmdline.txt'
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
BrokenPipeError: [Errno 32] Broken pipe
Traceback (most recent call last):
  File "/config/Airiana/./install.py", line 124, in <module>
    setUart()
  File "/config/Airiana/./install.py", line 23, in setUart
    boot_file = open("/boot/config.txt", "r+")
FileNotFoundError: [Errno 2] No such file or directory: '/boot/config.txt'
➜  Airiana git:(master) ✗

I think i will screw up my homeassistant system without a step by step for a rasberry with HA-os install, i want to try more but going to work for 6 weeks. Dont want the misses to get mad because of a downed homeassistant. I will try this on a ubuntu vm first i think

BeamCtrl, 2023-01-10:
sounds like a plan, I can not say what is wrong, Im running my home assistant in a docker on top of my normal rasbian buster installation.
//D good luck

Ontrackx, 2023-01-10:
Is it possible to get this via the http://192.168.86.101:8123/hassio/store  Home Assistant Community Add-ons at some point?
Tried manually adding the repo but it complaines with https://github.com/BeamCtrl/Airiana is not a valid add-on repository
Probably dome dependencies missing and stuff. Im a end user who likes to fiddle until i get things to work. It stops there :)

BeamCtrl, 2023-01-10:
No, airiana is not an addon for home assistant

BeamCtrl, 2023-01-14:
Closed as not suprted.

closed 2023-01-14 by BeamCtrl