BeamCtrl/Airiana #32: Get into trouble when installing

Ontrackx, 2023-04-28:
PS C:\Windows\system32> cd..
PS C:\Windows> cd..
PS C:\> git clone https://github.com/BeamCtrl/Airiana
Cloning into 'Airiana'...
remote: Enumerating objects: 3786, done.
remote: Counting objects: 100% (868/868), done.
remote: Compressing objects: 100% (286/286), done.
remote: Total 3786 (delta 604), reused 835 (delta 581), pack-reused 2918
Receiving objects: 100% (3786/3786), 782.50 KiB | 702.00 KiB/s, done.

Resolving deltas: 100% (2593/2593), done.
PS C:\> python3 ./install.py
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
PS C:\> cd airiana
PS C:\airiana> python3 ./install.py
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
PS C:\airiana> python ./install.py
Traceback (most recent call last):
  File "C:\airiana\install.py", line 17, in <module>
    import pwd
ModuleNotFoundError: No module named 'pwd'
PS C:\airiana>

Is there any way to ignore the pwd error? i know this is a phyton problem, but ive tried on ubuntu also I should be able to run this on other than a dedicated raspberry pi yes?

BeamCtrl, 2023-04-28:
Ive never tried it on something other than Rpi, it may however work with some fiddling. if you install the requirements from the requirements.txt file you should be good to go without installing, The install script should maybe have a caveat to only use on linux/rpi since it assumes alot about the local filesystem and services. 

1. install the requirements manually
2. try to startup ./airiana-core.py and see how far you get.

are you trying to run it via a IAM module or wired via rs485?

//Daniel

Ontrackx, 2023-04-28:
IAM, will try more tomorrow. Thanks for the answers 😎 i really want to try
this because it seems like a very good project. 😎

BeamCtrl, 2023-10-07:
closing due to inactivity

closed 2023-10-07 by BeamCtrl