BeamCtrl/Airiana #33: Modifying minimum air flow

jdybvik, 2023-08-13:
Hi. Long time user and fan of Airiana - running VR400 and raspberry pi. I have tried to edit airiana-core.py in a similar manner as "DrSammie"  [(see previous issue)](https://github.com/BeamCtrl/Airiana/issues/27) in order to modify minimum airflow.

I edited:
Line 550 req.write_register(102, 60) #was 40

Regardless of this, the fan always defaults to: Supply:30 l/s,1710rpm  Extract:30 l/s,1560rpm

Any pointers to increase minimum air flow is much appreciated.

BeamCtrl, 2023-08-14:
Just to let you know, I'm taking a look, don't have alot of time tho. Might take a few days. //Daniel

BeamCtrl, 2023-08-19:
Give the lastest master a test flight, `31a45bc5` I do not have a non savecair unit anymore for testing and as no real test framework has been created/used in this project its hard to debug without a real system. Hope this helps with changing the low flow rates. 

you will need to manually change the values of self.ef_base to the value you desire ref. airiana-core.py:545 or a little down in the file depending on the model you are using.
GL
//Daniel

jdybvik, 2023-08-20:
Thanks a lot, really appraciate your continued efforts into Airiana! 31a45bc5 is succsessrful in changing low extract fan, but not the supply. I had to make sure Airiana-core.py also wrote supply fan onto the register. 

I did this by changing line 554
                # req.write_register(101,40) 	#read only

to
req.write_register(101, self.sf_base)

closed 2023-08-20 by jdybvik