BeamCtrl/Airiana #37: installation fails in latest bookworm raspbian

dcolens, 2023-12-23:
this is FYI, I tried installing on the latest Rapbian OS (bookworm), the installation fails for a couple of reasons:

1. /boot/config.txt and cmdline moved under /boot/firmware/ in bookworm this breaks the install script that
2. it is no longer possible to run `pip install` outside a venv (see https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi)

**To Reproduce**
Steps to reproduce the behavior:
1. install bookworm on RPI
2. clone airiana repo
3. run python3 install.py

**tested on**
 - Device: RPI3
 - OS: bookworm

dcolens, 2023-12-23:
I managed to install, I made the following changes:
1. updated path for config.txt and cmdline.txt (/boot/firmware)
2. create venv in airiana's folder
3. installed deps using pip install (including numpy), because the pip install --user fails
4. `sudo apt-get install libopenblas-dev` (required for numpy to load properly)
5. create latlong.json file in airiana folder (without it, it would not start)

I think that's about it, hope this is useful

BeamCtrl, 2023-12-26:
Thank you for a detailed bug report. =). and solutions. WIll take a look on doing an update, maybe even a rework of the install scripts. Its been growing into some different kind of beast for a long time.

BeamCtrl, 2023-12-26:
Hope you got you installation working and that it works well, always fun to see that ppl are using it =)

dcolens, 2024-01-08:
yes, my installation is operational thanks for the great work :-)

BeamCtrl, 2024-06-17:
brief update, now the there is a branch called merge/autohotspot, it includes bookworm compat, and  rework of the installation script.

BeamCtrl, 2025-02-11:
resolved as of merge with auto-hotspot branch, 0610db09

closed 2025-02-11 by BeamCtrl