BeamCtrl/Airiana #38: JSON formatting error

jdybvik, 2025-03-31:
Long time Airiana-fan, again thanks a lot for developing and maintaining this gem. I´m testing the MQTT integration with homeassistant.

**Describe the bug**
JSON air.out prints np.floats which lead to JSON formatting errors on VR400. 

{"extract": np.float64(22.347665447297377), "coolingMode": "false", "supply": np.float64(18.37694805194805), "sf": 40, "ef": 50, "exhaust": np.float64(12.686525974025974), "autoON": "true", "shower": "false", "rotorSpeed": 100, "sfRPM": 1890, "energyXfer": np.float64(711.8354357204744), "efficiency": np.float64(77.9678251356542), "efRPM": 1860, "name": "VR400", "filterPercentRemaining": 15.9, "pressure": 1004.06, "filterInstalledDays": 313, "rotorActive": "yes", "elecHeater": 0, "inlet": np.float64(4.328371753246754), "electricPower": 81.85186413186477, "electricPowerTotal": 2.18, "referenceHumidity": 30.8, "calculatedHumidity": 39.4}

**To Reproduce**
Steps to reproduce the behavior:
Run Airiana 

**Expected behavior**
No np.floats in JSON out

BeamCtrl, 2025-03-31:
hope you did not wait to long ;) I put up a fix in  commit e189c8e6, "./update" to get it.
out of curiosity because I could not replicate the issue do you know what version of numpy you are using?
Please verify the functionality before I close the issue. =)

jdybvik, 2025-03-31:
Wow, not expecting a new commit this soon! I´m running numpy 1.24.2. 
Unfortunately I´m stil getting the same JSON output: 

{"extract": np.float64(22.312746163444476), "coolingMode": "false", "supply": np.float64(18.36300448430493), "sf": 40, "ef": 50, "exhaust": np.float64(12.227354260089685), "autoON": "true", "shower": "false", "rotorSpeed": 100, "sfRPM": 1860, "energyXfer": np.float64(691.9015537321318), "efficiency": np.float64(78.22042692793133), "efRPM": 1830, "name": "VR400", "filterPercentRemaining": 15.9, "pressure": 1003.36, "filterInstalledDays": 313, "rotorActive": "no", "elecHeater": 0, "inlet": np.float64(4.173773542600896), "electricPower": 68.8075169536382, "electricPowerTotal": 2.23, "referenceHumidity": 30.9, "calculatedHumidity": 39.6}

BeamCtrl, 2025-03-31:
ahh I see, seems there is an issue with newer numpys in that the __format__() method uses __repr__() to print to f-strings, this puts my fast solution to shame. But never don't give up I say. Gave it another go in 1be014c2.

And also, tnx for the kind words on using it =), always makes me happy that all that time spent is used somewhere =).

jdybvik, 2025-03-31:
amazing work, once again - no problems with json output now. Thank you very much! 

ref: 
{"extract": "22.338136879362356", "coolingMode": "false", "supply": "18.61034482758621", "sf": 40, "ef": 50, "exhaust": "11.84367816091954", "autoON": "true", "shower": "false", "rotorSpeed": 100, "sfRPM": 1860, "energyXfer": "656.1459542541935", "efficiency": "79.46208292110039", "efRPM": 1860, "name": "VR400", "filterPercentRemaining": 15.9, "pressure": 1002.9599999999999, "filterInstalledDays": 313, "rotorActive": "no", "elecHeater": 0, "inlet": "4.192003831417624", "electricPower": 70.30649141231046, "electricPowerTotal": 0.85, "referenceHumidity": 30.7, "calculatedHumidity": 39.7}

closed 2025-03-31 by jdybvik

BeamCtrl, 2025-03-31:
Excellent 🎉🥳

Skickat från Outlook för iOS<https://aka.ms/o0ukef>