BeamCtrl/Airiana #39: bug in install.py after fixes to ticket #38

DrSammie, 2025-04-15:
Hi,
Thanks again for a great program! found small bug.

**Describe the bug**
fails to install after last update.

line 213 in install.py the indent is tab instead of spaces causing it to fail. 

changing from
"	print("Setting up virtual environment...")"
to
"        print("Setting up virtual environment...")"
makes it work again.

with regards
Samuel

BeamCtrl, 2025-04-15:
Hi, thank you for the find and the kind words., it seems I did not test enough :) Fixed the issue, and added a formater to my hooks so as to avoid this happening again.  Fixed as of #6eaa8d4e, will leave it running for a while before closing the ticket as the formater fixed a whole lot of thing I have been putting off, thank your for getting me over the edge to actually getting this in.

BeamCtrl, 2025-04-27:
closed, after two weeks of ok operation.

closed 2025-04-27 by BeamCtrl