BeamCtrl/Airiana #43: YAML dependency not installed on fresh installation

johanberglin, 2025-05-05:
**Describe the bug**
Fresh install on RPI zero W, with bookworm 32bit Desktop
ssh to RPI as pi user (got another issue if not using pi as user) 
sudo apt-get updates
sudo apt-get upgrade
in home directory
git clone https://github.com/BeamCtrl/Airiana
cd Airiana
python3 ./install.py
after reboot no output to RAM directory

To solve problem:
manually run 
/home/pi/Airiana/venv/bin/python /home/pi/Airiana/airiana-core.py
got error about YAML module missing

cd /venv/bin
./pip install PyYAML

run
/home/pi/Airiana/venv/bin/python /home/pi/Airiana/airiana-core.py
result no yaml error

BeamCtrl, 2025-05-05:
Great find, looks like when I added the yaml configuration system I added the dependency to requirements.txt but missed the listing in install.py. fixed it right away,  very nicely written bug report. Im leaving this open untill I make sure the installer reads the requirements.txt instead of maintainging deps listsings in two places.

johanberglin, 2025-05-05:
Airiana har funkat mycket bra i flera år.
För ett par månader sen slutade det fungera, jag felsökte gav upp och gjorde en ominstallation.
Min RPI är headless så jag är beroende av fel loggar.
Vore bra om instalallationen och första uppstart loggade till filer som inte är i RAM.
Nu får jag checksum fel i kommunikationen med FPX.

Startat om aggregatet och min RPI flera gånger.

GNU nano 7.2                        err                                 Using /dev/Serial0      Mon May  5 14:49:58 2025                          Traceback (most recent call last):                                          File "/home/pi/Airiana/./forcast2.0.py", line 100, in <module>              f = open("latlong.json")                                                      ^^^^^^^^^^^^^^^^^^^^                                              FileNotFoundError: [Errno 2] No such file or directory: 'latlong.json'    Using /dev/Serial0      Mon May  5 14:50:28 2025
Traceback (most recent call last):
  File "/home/pi/Airiana/request.py", line 213, in modbusregister             self.response = self.client.read_register(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^                              File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    returnvalue = self._generic_command(                                                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    payload_from_slave = self._perform_command(functioncode, payload_to_s>                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>  File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    payload_from_slave = _extract_payload(                                                         ^^^^^^^^^^^^^^^^^                                  File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    raise InvalidResponseError(text)                                      minimalmodbus.InvalidResponseError: Checksum error in rtu mode: b'\x00\x0>During handling of the above exception, another exception occurred:       Traceback (most recent call last):                                          File "/home/pi/Airiana/request.py", line 213, in modbusregister             self.response = self.client.read_register(                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^                              File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    returnvalue = self._generic_command(                                                    ^^^^^^^^^^^^^^^^^^^^^^                                    File "/home/pi/Airiana/venv/lib/python3.11/site-packages/minimalmodbus.>    payload_from_slave = self._perform_command(functioncode, payload_to_s>
^G Help       ^O Write Out  ^W Where Is   ^K Cut        ^T Execute
^X Exit       ^R Read File  ^\ Replace    ^U Paste      ^J Justify

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

BeamCtrl, 2025-05-05:
Tjena, kul att den fungerat länge =), trist  att det strular nu. Ska se om jag kan hjälpa dig på plats och att vi fixar några problem påvägen. Initialt problem uppstod nog iom att jag stoppade in lite ändringar som inte de automatiserade testerna fångade. men så är det när man saknar testriggar. 

Jag funderar på om problemet du har härhör från inställningar på aggregatet. Har du checkat baudrate 19200 och no parity, one stop bit? jag har varit med om att mitt eget aggregat tappar inställningen när man gör den spänningslös.

Jag ser även att den klagar på att forcast saknar latlong, kör du den som WiFi AP eller har den Internet? den ska estimera position med ip geoloc men det verkar inte ha lyckats. Den kommer försöka igen men underlättar min förståelse om du delar med dig om lite hur det ser ut. Lagra loggar ej i Ram görs bara för updateringar(update.log) så absolut en bra ide att lägga det persitent, för installationen iaf.

Mvh //Daniel

closed 2025-05-07 by johanberglin